ExpressLRS/ExpressLRS #1: ExpressLRS

SZ-MXK, 2019-07-04:
Hi，
My FC has integrated ESP32. Are you interested in cooperating?

AlessandroAU, 2019-07-04:
Sure, we can have a chat. Join on discord? 
https://discord.gg/dS6ReFY

SZ-MXK, 2019-07-05:
I can't use discord，Do you use Facebook? +86 18818594513

AlessandroAU, 2019-07-12:
Did some testing of the system on the weekend, still have some issues. Much shorter than expected range. Not sure why. Perhaps preamble too short.

SZ-MXK, 2019-07-15:
https://github.com/RavenLRS/raven I've tested this LRS, which can reach 5 kilometers. If the distance is short, you may need to redesign the hardware

AlessandroAU, 2019-07-15:
I don't think it is a hardware problem, I am using self contained modules for the radio interface.

AlessandroAU, 2019-10-22:
Range issues fixed

closed 2019-10-22 by AlessandroAU