ExpressLRS/ExpressLRS #5: EU compatibility

stylesuxx, 2019-10-22:
Hey, interesting project - thanks for sharing.
In order to use it over here in the EU we would need this to run on 868MHz, which I assume would not be a big deal. How about LBT - did you consider implementing that at all?

AlessandroAU, 2019-10-22:
Do you have a reference for the EU lbt requirements? It doesn't take long to take an RSSI measurement of an active channel and in theory it could be done after each tx while the RX is busy

stylesuxx, 2019-10-23:
More information about LBT can be found here ->
https://www.etsi.org/deliver/etsi_en/300300_300399/300328/01.08.01_30/en_300328v010801v.pdf
see
section 4.3 for more details.

AlessandroAU, 2020-02-12:
Just had another look at this, freq band seems really really narrow. Not sure how XF and R9 handle it.

AlessandroAU, 2020-04-02:
868 bands have now been added, but LBT is not yet in use.

stylesuxx, 2020-04-02:
Awesome, thank you. Will order parts and start testing!

0crap, 2021-01-10:
Would love to see a 2.4GHz LBT version as well.
Parts are on order to make my first ExpressLRS 2.4G TX module.
Thx!

greenbigfrog, 2021-01-11:
Gonna document my research into this here.
(these results are based on a perspective from Germany. Different regions might still have different regulations, since most EU stuff still needs to be copied into local law.)

**2.4**:
- 10mW per channel
- LBT etc as defined in [ETSI 300 328-v1.8.1](https://www.etsi.org/deliver/etsi_en/300300_300399/300328/01.08.01_60/en_300328v010801p.pdf)

**868**:
- try minimizing noise, disturbance of self and others, etc [2014/53/EU](https://eur-lex.europa.eu/legal-content/en/TXT/?uri=CELEX:32014L0053) (implemented in  [FuAG](https://www.gesetze-im-internet.de/fuag/FuAG.pdf))
- different power/bandwidth/duty cycle limits across multiple channels as outlined [Allgemeinzuteilungen SRD  2018/05](https://www.bundesnetzagentur.de/SharedDocs/Downloads/DE/Sachgebiete/Telekommunikation/Unternehmen_Institutionen/Frequenzen/Allgemeinzuteilungen/2018_05_SRD_pdf.pdf?__blob=publicationFile&v=6)

From what I've been able to find, there's no actual on paper requirement for LBT on 868. [ERC-REC-70-03E](https://www.arcep.fr/fileadmin/reprise/dossiers/frequences/ERC-REC-70-03E-version02.PDF) exists, but only is a recommendation, and was only implemented in france law as such from what I've been able to find.

greenbigfrog, 2021-01-14:
Ok, so I've spent a few more hours reading the EU directives regarding 868. (These also (partially) apply to 2.4)

[2014/53/EU](https://eur-lex.europa.eu/legal-content/en/TXT/?uri=CELEX:32014L0053) is about radio equipment. The only truly relevant part of it is Article 3.
[removed quote]
I think we fulfill all of these already to a sufficient degree (for 868).

Actually had to delete most of my comment again, since I just discovered ETSI EN 300 220
This is a jungle... I guess I've got more reading to do

0crap, 2021-01-14:
>  
> Actually had to delete most of my comment again, since I just discovered ETSI EN 300 220
> This is a jungle... I guess I've got more reading to do

The header of that document says "Short Range Devices (SRD) operating
in the frequency range 25 MHz to 1 000 MHz;" so I guess that only applies to 868 MHz and not 2.4 GHz.

greenbigfrog, 2021-01-14:
> The header of that [document](https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_30/en_30022002v030201v.pdf) says "Short Range Devices (SRD) operating
> in the frequency range 25 MHz [...]

Yes, sorry I didn't mention that more explicitly.
`ETSI 300 328-v1.8.1` is the relevant document for 2.4, and ETSI EN 300 220 for 868 (SRD frequencies (would also include 433)). (be aware, ETSI EN 300 220 has multiple parts, and I'm not sure how much of it is actually finalized or how they exactly are split up.)

ManoDaSilva, 2021-01-20:
Perhaps we could also consider 433MHz?

greenbigfrog, 2021-01-20:
> Perhaps we could also consider 433MHz?

433 has even less "channels" from my understanding. It's not very interesting like at all, especially if you want to fly with multiple people on 433.

That being said, the requirements should be rather similar to those of 868. But feel free to share whatever you find out or know

0crap, 2021-02-16:
> Gonna document my research into this here.
> (these results are based on a perspective from Germany. Different regions might still have different regulations, since most EU stuff still needs to be [...]

I've found a bit more recent ETSI document [ETSI 300 328-v2.2.2] (https://www.etsi.org/deliver/etsi_en/300300_300399/300328/02.02.02_60/en_300328v020202p.pdf)

Also this document from TI which is a bit more readable, regarding CE regulations, might help with the 2.4GHz LBT stuff.
[CE Regulations for SRDs Operating in License-Free 2.4 GHz/5 GHz Bands](https://www.ti.com/lit/pdf/swra670)

diego0815, 2021-02-28:
Here the regulations for Switzerland which are pretty much tracking EU requirements:
https://www.ofcomnet.ch/api/RIR/1008/09

More frequencies: https://www.ofcomnet.ch/api/RIR/1008

2.4GHz data transmission: https://www.ofcomnet.ch/api/RIR/1010/01

Or the more restrictive version:
https://www.ofcomnet.ch/api/RIR/1008/11

Seems like on 869MHz you have the choice of either doing LBT or staying below 10% duty cycle.

For 2.4GHz FHSS with minimum 100kHz bandwidth allows for 100mW, else you have to do extreme wideband to reach 100mW. LBT is one possible means to achieve coexistence.

Dropsleutel, 2021-03-01:
I have some feeling adding more hopping channels so they overlap each other may possibly be better reject, some small band other signals, also we are sure to use the maximum available space in the band. the chance that two systems hop onto each other partially should stay the same, full hit will become smaller.  It seems crossfire use approx 3 times more channels than would fit the band ?

diego0815, 2021-03-01:
Cept regulations are here: 
For generic short range devices: https://efis.cept.org/adhoc_grabber.jsp?annex=4

And for wideband data transmission: https://efis.cept.org/adhoc_grabber.jsp?annex=6

Actually pretty identical to the swiss documents above, as those are a national implementation of the cept rules.

d-hartmann, 2021-06-08:
Any update ob the EU compatibility? Is there any possibility to support here in any way?

Creat, 2021-08-02:
I'm also very much interested in the state of LBT for 2.4 GHz. Now that 1.0 is released, does it include support for those regulatory requirements? Or is it on the roadmap?

DocTrucker, 2021-09-07:
Will the LBT requirement restrict the performance of ELRS at all, hense the difficulty in implementation?

As radio manufacturers have had to comply with EU requirements such as LBT since 2015 may this mean the current youtube tests be a little misleading?

FlydayOne, 2021-11-15:
Hello!
I would really like to use 2.4 GHz ELRS in the EU. LEGALLY!
My understanding is, that this is not possible at the moment! Is this correct?
If I'm correct, will there be somewhen a "legal" version that fulfill EU regulation requirements?
And if this is planed, is there a rough idea when this could be?
As this issue was open a long time ago, I have the fear this is not in the focus. But I believe this would be a real benefit for the link as it open it up to a big amount of users that now don't use it for exactly this reason.
Thank you and best regards

avsaase, 2021-11-15:
As I understand it, only implementing LBT doesn't make it legal. The hardware/software combination would also need to be certified and there is no way the manufacturers of the hardware we have now are going to bother with that. Hell, they should just slap on a fake CE/FCC sticker like they do with all their other rf hardware.

FlydayOne, 2021-11-15:
Actually I do agree but only partly.
Whether or not LBT is needed is out of my knowledge!
But for the rest, I believe the odds that a manufacturer is going through a validation process is by far higher if you offer them a solution that would pass the process than if you don't. 
I am currently using ImmersionRC Ghost JR Module with Ghost! And I am quite sure this link complies. Or I am wrong?!?
Why shouldn't another supplier offer an EU version of such a module if he can get the software for free and the hardware already exists?
I believe that this would be in the interest of the manufacturer (potential better market access), all the manufactures of ELRS components and ready RC-MODLES (more potential customers) and the community (obviously).

stylesuxx, 2021-11-15:
@FlydayOne CE certification would be necessary for each and every receiver and transmitter. In order to pass the CE certification LBT would need to be implemented (among other things) and the manufacturers would need to go through certification with every device. Since you have to pay for the certification (or at least for all necessary test), I can not imagine that manufacturers would follow through with this.

This is also the reason why for example TBS or IRC don't release a new product every other day, they release something go through the certification and then run for the product as long as they can. If you take a look at - for example - the BetaFPV product line, they have around 8 items that they would need to get certified, and I don't see this happening anytime soon.

Just slapping the CE/FCC label on it as @avsaase suggests is not the best idea either since it might lead to BIG problems on the vendor site. If for example I import something to the EU with the purpose of reselling I am responsible to make sure it meets the standards, so if it is a fake CE certification, and I just assume that the manufacturer actually did the certification and he did not, I - as the importer am liable for that.

So yeah, unfortunately I don't see any way that ERLS is going to be legally usable in the EU anytime soon.

avsaase, 2021-11-15:
> Just slapping the CE/FCC label on it as @avsaase suggests is not the best idea either since it might lead to BIG problems on the vendor site. If for example I import something to the EU with the [...]

I wasn't really serious about this as any sort of solution but I've heard that this happens with some RF hardware in our hobby. Surely, the same risks for importers exist when there is no CE label at all, but I can still find ExpressLRS hardware in local rc shops. I'm not saying I'm opposed to doing things the right way - these regulations exist for good reason - but looking at the way things work right now the lack of LBT/certification doesn't seem to be a problem for manufacturers or users. Maybe there is an untapped market of law-abiding citizens that also never use R9 flex firmware and would like to switch to ELRS, but I don't see manufacturers taking the steps to cater to these users.

FlydayOne, 2021-11-15:
Thank you! This is not the answer I was hoping for but it is a clear answer. 
I still hope there will be somewhen a legal way to use ELRS in the EU.
So sorry... but for now I will stick with 👻.

avsaase, 2021-11-15:
You can flash ELRS to your ghost firmware, that way you still have your sticker and no one will ever know :stuck_out_tongue_winking_eye:.

stylesuxx, 2021-11-15:
@FlydayOne please take my answer with a grain of salt. I opened this issue two years back soon after I found this project. At this time I was toying with the idea to get a small series produced and sell it over here. I am not affiliated with the project itself, so it might be, that the team has something worked out with a manufacturer who is willing to go through the whole certification process. So don't misinterpret my answer as an official statement.

@avsaase I don't think that too many people in the hobby here in the EU care too much about flying legally all the time. Don't get me wrong, most people - if not all - that I know here are very responsible pilots, but expecting them to - for example - only fly on 25mW video output in a Bando is a bit naive. But sometimes you need to be legal, for example when attending races. Often times only certain video transmitter models are permitted. If you don't follow those rules, you will be excluded from the race. Which makes sense: As an organizer I don't want to risk it doing something that is against the law and might cause trouble for myself, and the bigger the event, the bigger the chance of something happening and being exposed.

So yeah, as long as you are flying for yourself and are doing so responsibly, the chances of anyone catching you with "illegal" hardware are basically non existent...

This is just my point of view and not an encouragement to do anything illegal.

FlydayOne, 2021-11-15:
I can understand all the mentioned opinion.
Still, I believe it is making a different wether or not the product would pass a potential validation process.

@stylesuxx 
You said you were toying with the idea to producing a series... maybe someone else, maybe even an established manufacturer, as well? As long as the software doesn't comply, noone will find out!
That was the reason for my question!

Additionally I believe everybody can judge on his own what is the risk of using a not authorized link and if it is meaningful not to comply with the standards.
Personally I wouldn't in the actual situation.

But I would feel more comfortable with a link that generally complies but has no certificate rather than with one that doesn't. 
And I would feel even more comfortable with a certified tx but a non certified rx e.g in a BNF.

sherghan, 2021-11-18:
Please correct me if I'm wrong. I'm expert in neither law nor radio equipment. Just an RC hobbyist. 

My understanding of our EU law is as follows:
* It's Me as a user of the radio equipment, who is responsible for it's compliance with the regulations.
* If manufacturer puts a certificate sign on the equipment it means: _I say it's compliant, trust me_ - this can be some form of excuse for me
* If I know for sure that equipment is compliant, I don't need a certificate on it. Though it's welcomed and would be helpful for all those who just want to buy without digging into details.

Still to be safe and calm we would need to know that ELRS drives the hardware in a compliant manner at first.

I know it's tempting to hide "below the radar", but it is in fact a ticking bomb. And probability that someone will try to shot us down is growing along side with popularity of ELRS.

Betriebsrat, 2021-11-18:
Yea basically what sherghan said. While having a CE sign makes you able to have the manufacturer in charge, it is not the case that a missing CE makes it illigal per se aa far as my research went.
So if lrs speaks the right frequency with lbt we would be able to legally fly that in eu. The downside is that if shit hits the fan you have to prove somrhow that your rc link was eu compliant.
+1 for having LBT still.

stylesuxx, 2021-11-18:
@sherghan ultimately you are responsible in the first place for your actions, yes. BUT - as I stated before - the importer of the product is responsible for what he imports and sells, if it does not comply with the law, he will be held responsible this is pretty clearly stated [here](https://ec.europa.eu/growth/single-market/ce-marking/importers-and-distributors_en). So this makes it very risky for anyone who wants to sell ELRS hardware in the EU.

@Betriebsrat output power is also one of the limitations that needs to be kept in mind.

From what I understand you are only allowed to use compliant devices, see for example [here](https://www.traficom.fi/en/communications/communications-networks/conformity-radio-equipment-guarantees-interference-free):
> ...radio equipment must meet the applicable requirements in the EU. Only compliant radio equipment may be sold and used in the EU.

More information about how to acquire the CE marking: https://europa.eu/youreurope/business/product-requirements/labels-markings/ce-marking/index_en.htm
Radio Equipment Directive (RED): https://ec.europa.eu/growth/sectors/electrical-and-electronic-engineering-industries-eei/radio-equipment-directive-red_en

Dropsleutel, 2021-11-18:
I think the hardware design and chipset etc it is likely compliant, especially when there is a bpf at the output beween antenna.

So it could pass all tests for spurious interferences etc if needed.

However at the moment it is not compliant for power levels and hopping sequence. There must be a locked 25mW  lbt mode.

If so probably a CE label can be stocked on (declaration of conformity). 
It could be backed up even by a test if happymodel etc whatever manufacturar want to proove and be sure. It can than be officially legally sold in EU. And Authorities could confiscate your radio and test it and it should pass. Also you'll be insured etc.

But Testing a by design non compliant system, makes no sence.

modifying the system afterwards and flying it, by the user, it is not for the seller/manufacturar responsibility.

FlydayOne, 2021-11-18:
I go with this statement except of the power level. It seems that  with hopping somehow 100mW is possible. Same as GHOST or TRACER already do.

ChrisOhara57, 2021-12-04:
Hello
Just some information to the complexity of making a legal use possible.
(From my limited personal knowledge)

1. The product has to be in the specification as law requires.
2. All the required test have to be done for every type and every manufacturer that is producing this type.
   (Some products like cars do even need a different aproval for every factory they are produced in (even when the parst are the same))
3. The results and parts have to be handed to a certified laboratory (the laboratory itself has to be certified by a European authority)
4. The laboratory has the right to write a certificate and defines a unique CE-number.
   (This does only declare that the the meassurements of the product is comform to EU-law. 
    It does not even say that the laboratory has confirmed by his own meassurements.)
5. Now the product can be signed with CE.
6. Someone is importing this product to Europe.
   The law says: I can not get hold of someone outside the EU.
   So the one who does import this is only allowed to put this in circulation if he sells it with A Letter/Certificate of Conformity.
   It has to be signed by someone here in Europe who declares that he is fully responsible that the product is legal.
   (Example:  I found Taranis QX7 at Voelkner.de where the COC is signed by someone of Conrad Electronic.
              I found Taranis QX7 at EngeMT.de where the COC is signed by someone of Engel Modelltechnik.
              I`m sure they both hold track of the SNr of their imported products)
   If a private person imports a product he is fully responsible (in front of a judge) that it is in the specification the law requires.
   The cerificate itself does not really help (One of the first statements there normaly is that the laboratory 
   has only checked that the given meassurements are inside the required limits but they are not responsible for the product as such.)

Now there is the option to build a modul/receiver yourself.
In this case you have to do all the meassurements and then you to the responsible authorities and apply for a operating licence.
The fact that modul, receiver and firmware may fullfill the technical spezifications alone is not sufficient.
(In the EU you can compare this with the followiong example:
  You can build your own car in your garage and it may be technically to legal specifications.
  But your are not allowed to drive on the street as long as you do not get an offical operating licence for this car.)
Even a change in firmware normaly requires a new approval.

These international standarts are equivalent (EU/CE similar to USA/FCC similar to China/CCC)
It is working vice versa.
So even inside China you have to obey the CCC regulations which normally are a 90% copy of European or US regulations.
Besides there are also many other "independent" countries not belonging to one of the three groups.
But normally they adopt to one of the three (example: Switzerland and some other non EU-members are in fact similar to EU)

My personal conclusion: 
    If ExpressLRS shall become something more than an experimental project you need companies that are
    willing to produce and import the products on a legal basis. Otherwise it is only a matter of time till
    responsible authorities "keep an eye on it".

DocTrucker, 2021-12-06:
I think this topic is being pulled a little off topic. Most folk realise this can't be CE marked unless a company does the legwork but the absence of look-before-talk (LBT) is a big missing feature that prevents compliance with the regualtion unless you choose to run at 10mW or less. Until look before talk is in the default shipping of ExpressLRS there is no hope of a company legally selling ExpressLRS products in the UK, let alone using them.

Lets use this topic to flag clear breaches - such as no LBT - so these can be resolved when practical and significantly increase the chance of a manufacturer coming along and supporting the product. With clear breaches identified and resolved when practical a manufacturer is far more likely to want to produce the product as the market will include EU/UK too.

Perhaps things are subtly different for radio products to some products covered by the machinery directive but my experience of CE certification is that it is often of little value (outside key product areas as said before) - but still a legal requirement. It is largely a self-certification process unless dealing with key product area such as things like working in or containing explosive atmospheres the CE mark. Not all products need a unique externally verified CE number.  The certificate merely identifies the people/company that say that the product conforms to the required regulations, often citing what standards the product has been built to in order to achieve conformance with the regulation. However, if there was a product shipped with a CE mark and the Health and Safety executive in an EU/UK requested the supporting documentation following an issue (aside from certain product types the company gets some time to pull together the documents) and this was found to be inadequate then there can be legal consequences.

I've seen many products - especially cheap 3D Printers - whose build quality and risk management is obviously lacking. The signatories for these are often minimal value import specialist companies that could disappear quite easily if there were a significant issue.

ChrisOhara57, 2021-12-06:
Hello
Sorry, i didn't want to push. i only did want to help.
I worked at a carmaker and had trouble in my daily work because engineers offen thought it would be sufficent that the product meets the specification of the law forgetting that there also procedures required by law.
As an example that radio concerns it is the "**Guide to the Radio Equipment Directive 2014/53/EU"**  (sh. upload).
So you may think a car is not a radio but in fact there are a lot of small radio in it (like the key) and everyone of then requires it own certiftication and its own markings and everyone of the part hast to be listed with identification number etc. in the printed manual in the car. (really shit)
Did nearly break down the complete car industry.
In this directive you find the a lot of things required as procedure and responsibilities.
I do not know whether there others because I am not an electronic man.
I even haven't read it all.

Most interesting parts starts at page 20 (manufacturer, importer, distributor)

There is also this sentence: 
"Where the radio equipment is capable of taking different configurations, the conformity assessment shall confirm whether the radio equipment meets the essential requirements set out in Article 3 in all possible configurations."

So if a radio equipment 
 has configurations in that can be switch on by the user and are not allowed in this region the part fails the conformity assessment.

As I said before  I only wanted to help.
It is a great project. Wish you luck.
Will leave now.

regards Chris

[20181219 RED guide.pdf](https://github.com/ExpressLRS/ExpressLRS/files/7660850/20181219.RED.guide.pdf)

DocTrucker, 2021-12-06:
Hello Chris, I think my response came over wrong. Wasn't a go at yourself.

You highlight a good point there about a user being able to change the configuration. In the user safety risk assessments on the machines I worked on we had to draw the line somewhere. So for us an end user could defeat laser safeties on the machine - rendering them inoperable - if they had interlock defeat keys available from many sources. Basically the if tools were required to defeat the interlocks then the actions after that were intentional actions beyond the manufacturers control, and thus beyond their scope of liability. This would be comparable to Joe Blogs taking his family car to a tuning garage and mapping the engine to the point where it failed emissions tests (side stepping VWs diesel gate here! :-D.

So for any manufacturer supplying Express LRS gear their liability would stop if the user flashed non-approved firmware. Without that protection many existing bits of radio hardware that is CE marked, would no longer be CE markable because the user could flash their own, or even ExpressLRS firmware onto it. So, the only requirement here would be to have a flag at the compile stage of the firmware that locks out all non compliant otions for EU/UK/US/Whatever territories. This is how the OpenTX is dealt with and I believe there are examples of Radio hardware with CE marks and OpenTX based firmware.

ChrisOhara57, 2021-12-06:
Yes I think you got it.

**A good example** how it works are the genuine radios and modules of Frsky.
Same hardware but delivered with different firmware (ECE, FCC). Even when I flash OTX firmware FCC on an EU radio and the FCC firmware on the module it would not work. I would need an extra LUA script (which a normal dealer will not give to me) to change a special bit and make it work. (A friend of mine got a wrong version delivered by a Chinese dealer and he could flash OTX and module-firmware with ECE but it didn't work without the technical support of FRSKY and a special lua-script)
So the radio tested under all the configuration  a user can change (even by flashing) on his own does pass the required test and therefore it has the CE-marking.
On the receiver side Frsky (Access protocol) has only one unique firmware. That is ok because the transmitter module defines the mode it shall work.

**Against regulations:** Multiprotokoll 4in1 modules. You do not find any of those with a CE marking.
There are protocols (examples: Frsky-V8, D8 or Spektrum-DSM2 ) which (not having LBT) are not allowed to put into circulation since 2015 . (Mind you the receiver are still legal to sell cause they (having no telemetry) are not emitting.)
Because the user can switch to these protocols the conformity test would fail.
Having no CE no serious German dealer sells this modules. (You could find them in ebayso. But normally these few are modules delivered directly from china or small dealers who do sell these articles as a side job.)

**Cheating while not really to regulations**. Radios with 4in1 Multiprotokoll module.
The one I checked did this with a trick. In the certificate the laboratory stated that the measurements the maker gave them are all to EU specification (no own tests were made at the laboratory). The problem is, that there where no measurements of the forbidden protocols in the papers. But they can easily be selected by the user and are also advertised. And that is clearly against the rules.

And you are also right if a user flashes an other firmware it can get illegal (User problem). So there has to be an approval of conformity for a new firmware. That's why you seldom see major new firmware for the emitting parts of the Frsky products (if they can avoid it). Same for every other serious RC Radio company.

Mind you that if you read more of the paper I had attached you will also find that there definitely has to be a CE marking on radio products (or papers) and that importer and dealer has to make sure that these exist together with certification and letter of conformity. In the case of persons buying modules at Banggood etc the buyer becomes the importer.
Hobbyists (like me) who are organized in clubs are aware of the situation and many of us are cautious of getting the club and ourselves into trouble. (That's likely to be the reason why you do not see certain radios in our club)

Regards Chris

PS I'm really not aware of the FCC regulations but I doubt that there aren't any anlog procedures.
My personal experience is that US authorities do have strict laws, rely more on self certification but are then concentrating on prosecution. (Less officialdom but maximum on deterrence)

ChrisOhara57, 2021-12-07:
**Chinese market**
Just googled for the Chinese equivalent  which is necessary for **imported or local produced** radio equipment.
There are the technical standards (describing allowed radio power and frequencies etc.)
These have to be approved with a "SRRC" Type Approval for Radio transmitter (SRRC stands for State Radio Regulation of China).
With that certificate you can get the CCC certificate for the part.
And then you are allowed to mark the part and sell in China.
details explained see link

https://www.china-certification.com/en/srrc-type-approval-nal-type-approval/

MartinHugh, 2021-12-11:
+1 for LBT please.
I understand that an argument for not providing LBT is that the hardware is not CE compliant.  But it could be at somepoint. I see even FrSky are selling Express LRS recently under their Vantec brand.  
An option (run time or compile time) to enable/disable LBT would make this easier for those trying to remove obstacles to get this excellent project accepted as a legal system.

ChrisOhara57, 2021-12-11:
1.  "Other way round":The hardware is not CE compliant **because** it has no LBT (but there may be other issues as 
       well beside LBT. Very likely the power also has to be limited)
2.  But still if it was technically ok you will need a company that is certifying it and an importer (for a letter of compliance)
3.  I contacted HorusRC about CE/FCC certificate. Answer:  **No marks at present.**
     So legal use is not possible at the moment (neither in Europe nor in USA)
4.  Keep in mind neither Frsky nor HorusRC does offer ELRS or even MPM via  a warehouse outside China (nor does any
     dealer I know).
     So the risk is 100% yours.  (I confess that I would like to try the ELRS myself)

MartinHugh, 2021-12-11:
@ChrisOhara57 many thanks for your comments.

Alfinto, 2021-12-11:
isn't LBT allowed in FCC?
why not go LBT all the way world wide?
when two non LBT packets collide everyone losses, when two LBT TX wants to transmit at last one will transmit?

radiomaster just announced an ELRS TX (Zorro) with CE: " RF power: TBA (CE and FCC Version will be available)"
https://radiomasterrc.com/articleb-236.html

anyone an idea how they want to achieve that?

i currently fly my whoop with 10mw in my home and my 5" out at fields with no one around, that's somewhat acceptable for me in CE land.

but to fly at the club with bigger birds and helis, CE would be requirement, also PWM RX and integration in heli flight units.
(once you have jeti and spirit with their integration, its hard to like something else.)
 
but spirit is already working on CRSF support and spirit have already lua scripts for openTX, so only LBT is holding my helis back from ELRS.
( https://www.spirit-system.com/phpBB3/viewtopic.php?f=24&t=3757&start=40 )

ty for creating ELRS

krotow, 2021-12-22:
> isn't LBT allowed in FCC? why not go LBT all the way world wide? when two non LBT packets collide everyone losses, when two LBT TX wants to transmit at last one will transmit?
>
> radiomaster just [...]

They mean ELRS LoRa band here (868 MHz for EU). Nothing was said about LBT support.

> 900MHz actually is 868MHz (EU region, CE regulations) 

I also vote about proper LBT support in ELRS. By logical conclusion and as EU country citizen. It definitely will make ELRS even better.

sherghan, 2021-12-22:
> They mean ELRS LoRa band here (868 MHz for EU). Nothing was said about LBT support.

In the same table they specify _Operating frequency_ as "_2.400GHz-2.480GHz_", so it's not _868MHz_.

Also power is not _TBA_ anymore, it's "_Up to 250mw ( CE & FCC Versions will be available)_".
Sill no word about LBT though.
...I assume they will limit it to 100mW in EU.

> I also vote about proper LBT support in ELRS.

:+1:

krotow, 2021-12-22:
> In the same table they specify _Operating frequency_ as "_2.400GHz-2.480GHz_", so it's not _868MHz_.
> Also power is not _TBA_ anymore, it's "_Up to 250mw ( CE & FCC Versions will be available)_". [...]

Ouch! Mistook with ELRS description in Jumper T-16 manual (was open in next browser tab). Then they indeed mean output power. 100 mW is still 10 km - more than enough for my whooping and freestyle. DJI FPV link in my quad will vanish faster and I'm not sure are battery will be enough to come back :)

For long range pilots they can do the same as TBS did in Crossfire radio settings - "Open" region with a possibility to have output power above 100 mW.

maexf, 2021-12-22:
Plus one for LBT. Making ELRS accessable for European user would give a push for the community and make FPV more open for new user.

avsaase, 2021-12-22:
> anyone an idea how they want to achieve that?

I've heard they will do the CE certification with a firmware limit of 10mW since that doesn't require LBT. See https://github.com/ExpressLRS/ExpressLRS/pull/1161. I assume they will also ship it with this limit (at least to the EU).

sherghan, 2021-12-22:
They are willing to go through certification process even without LBT.

This should be a strong proof that coding LBT makes sens and that it will be used by producers to provide CE certified radios with decent capabilities.

MartinHugh, 2021-12-22:
I dont think they even need to ship it with the 10mW constraint. They only need to demonstrate that they have tested all the requirements of CE. If they state they have done this at 10mW then they get to skip CE.  I am not sure that helps retailers and users who want to stay legal.

Why on earth not just implement LBT and make it configurable from the LUA so end users have the option to fly with it or not. I know of a number of flying clubs which require insurance to be able to operate, demand that radios are legal in order that everyone can fly.  This project has achieved a lot in a short time.  I am struggling to understand what the problem is.

pkendall64, 2021-12-22:
Implementing LBT is not that simple! We have had many discussions about this in the development channel. The main problem is that you cannot talk on the channel if there is energy on that frequency over a certain threshold and the LoRa chip really only tells you if another LoRa chip is talking.

sherghan, 2021-12-22:
LoRa isn't the only 2.4Ghz radio on the board.
Would it be possible to use the WiFi/Bluetooth radio to run LBT check?

JyeSmith, 2021-12-22:
Wheres my LBT *stomps foot*

As Paul said, it looks like LBT is only possible with the sx1280 to detect other LoRa packets.  There does not appear to be a general rssidbm reading for other modulations e.g. FSK

I invite anyone to review the docs and find a way to do it https://www.mouser.com/datasheet/2/761/sx1280_81-1107808.pdf

Semtech have also been asked directly, but with no response.  ELRS has also asked manufacturers to ask Semtech, but with no answer.

I understand the need for CE with clubs and insurance.  CE can be obtained with 10mW and no LBT, which is fine for flying at club event.  Any freestyle or LR user I dont imagine is at a club or concerned about insurance.

But what about LBT?  From what I gather the rssidbm cutoff for LBt is -51.... -51!!!  You need to be sitting next to another tx for this to be an issue.  So the only person you are affecting is the other FPV guys standing right next to you.  Take a couple of steps apart and then the rssidbm threshold doesn't come into play and LBT is never used.  You can also see how this doesn't make a lick of difference to the receiver.

ChrisOhara57, 2021-12-22:
> I dont think they even need to ship it with the 10mW constraint. They only need to demonstrate that they have tested all the requirements of CE. If they state they have done this at 10mW then they get to skip CE. I am not sure that helps retailers and users who want to stay legal.
> 

This would be not legal: See following lines as part of the "Guide to the Radio Equipment Directive 2014/53/EU"
 (and this directive defines how to test for a RED certificate and you need to get CE)

"Where the radio equipment is capable of taking different configurations, the
conformity assessment shall confirm whether the radio equipment meets the essential
requirements set out in Article 3 in all possible configurations."

MartinHugh, 2021-12-22:
> This would be not legal: See following lines as part of the "Guide to the Radio Equipment Directive 2014/53/EU" (and this directive defines how to test for a RED certificate and you need to get CE) [...]

Fair enough Chris. Which begs the question how legal TBS gear is with their "open region" setting,

sherghan, 2021-12-22:
> Wheres my LBT _stomps foot_

I'm sorry to make it sound like this.
I really do appreciate the work that You do here and the freedom that we all get from it.

It's that I try to share and promote ELRS around and lack of certification is a barrier, especially for those with licenses and insurances.

JyeSmith, 2021-12-22:
Its not a question of if ELRS will do LBT, but just how to do it.  We have already put 10s of man hours into the problem, with the whole team of devs and talking to manufacturers.

Future discussion should just be technically about how to do it.  Anything else just become distracting noise.

Any reference to the documentation of how to do LBT, IoT implementations, or examples from Semtech are appreciated.

https://www.mouser.com/datasheet/2/761/sx1280_81-1107808.pdf

ChrisOhara57, 2021-12-22:
> 

> Fair enough Chris. Which begs the question how legal TBS gear is with their "open region" setting,

Sorry I do not know about that.
But I just looked at it at  the TBS Homepage.
Looks like a "hidden code" to activate something that is surely not mentioned in any official manual delivered with the hardware.
I think everyone has to draw his own conclusion on that.

In case you want to un-lock your TBS CROSSFIRE you have to enter the Konami Code. 
Power up your TBS Crossfire wait until you see the main screen then do the Joystick combination "up, up, down, down, left, right, left, right, enter enter". Got to the "Menu" -> "General" and you should see a new selection called "Region". Select "Open" to unlock your TBS CROSSFIRE.

MartinHugh, 2021-12-22:
Chris - this is only if you have selected a region in the first place
By default the region is already open. 
For world wide shippping I assume.

Users are "supposed" to set their own region.  Or not

FlydayOne, 2021-12-22:
To come back to the main point:It seems the willing of implementation is there! Thumbs up!👍 It is really nice to hear that this topic is treated with high effort! Thanks for that! It is your free time you spend for this!Please know I appreciate it a lot!The question is just how to do it!So, please, everyone that can help is appreciated!Without knowing anything about it, how does immersionrc (ghost) do it? What chipset is used in the tx module?I know they explicitly say to use LBT!So... how?Again, I do not know... anyone how does: please reply!

[127 more replies]

closed 2024-07-31 by github-actions[bot]