ExpressLRS/ExpressLRS #63: Implement usage of MSP protocol in ELRS

wvarty, 2020-04-03:
PR #61 adds support for the MSP protocol in ELRS. We need to define some requirements around how it will be used within the codebase.

Some considerations are:

* Which UART will MSP use to send/receive on? UART2 on R9M? Which one on ESP32?
* Will that UART be reserved for MSP, or multi-use? I.e.
  * will we still pump debug log msgs out the same UART?
  * If yes, how will that affect any parties on the other end of that UART that are expecting MSP replies? The MSP class does not require that it has sole control over a UART, so we can toggle who's using the spare UART in realtime.
* When do we want to be listening for MSP data and acting on it?
  * In the main loop() function via a Serial.available() call?
  * Will this affect/be affected by the interrupt processing on RF packets?
* What should the TX code do when it receives a complete MSP packet? We need to think about how we forward that to the RX, then to BF, and/or how we process it if the TX should act on it (i.e. SetRFRate packet)

wvarty, 2020-04-03:
UART1 on the R9M is inverted (so rate limited and trick to un-invert) so UART2 is a better choice. UART2 is spare on ESP32, so we should use this on both for MSP.

wvarty, 2020-04-03:
> When do we want to be listening for MSP data and acting on it?
> In the main loop() function via a Serial.available() call?

we can add a `processMSPPackets()` call to the main loop() and check for `Serial.avalable()` to send to MSP class.

JyeSmith, 2020-04-03:
>     * Which UART will MSP use to send/receive on? UART2 on R9M? Which one on ESP32?

Yep, I agree uart2 for the r9 and esp.

>     * Will that UART be reserved for MSP, or multi-use? I.e.

For now I think we just go single use, unless its super easy to make mulit-use.  I dont want a future use case that may never happen causing us problems right now.

>     * When do we want to be listening for MSP data and acting on it?
>      
>       * In the main loop() function via a Serial.available() call?

Im happy with this and for the MSP data to be acted on immediately e.g. change mode, send MSP packet to rx.  But ideally should be an interrupt eventually.  Currently I can only think of MSP packets that would be sent when the quad/craft isnt flying.  If this does become a critical issue maybe only accept MSP packets when aux1 is off e.g. craft not armed.

>       * Will this affect/be affected by the interrupt processing on RF packets?

Unsure.  Maybe spam the uart and check LQ?
 
>     * What should the TX code do when it receives a complete MSP packet? We need to think about how we forward that to the RX, then to BF, and/or how we process it if the TX should act on it (i.e. SetRFRate packet)

We have 2 spare RC packets so lets use one for MSP.  I think the Rx should send it to BF in place of the next linkstats packet.  The Tx should also act on it immediately.

The current use cases I can think of are, changing mode, tx power, and vtx channel.  All things that are generally done when grounded.

wvarty, 2020-04-03:
> Yep, I agree uart2 for the r9 and esp.
Done. UART2 it is.

> For now I think we just go single use, unless its super easy to make mulit-use. I dont want a future use case that may never happen causing us problems right now.
It actually is super easy to do multi use with the lib. You just send the lib bytes (it doesn't know / care where they came from), and for sending, it pipe to a `Stream*` object which gets passed in as part of the function call to do the send, so we can still send log messages etc. in the background if we want without writing any extra code.

> Im happy with this and for the MSP data to be acted on immediately e.g. change mode, send MSP packet to rx. But ideally should be an interrupt eventually. Currently I can only think of MSP packets [...]
Let's go with immediate action in `loop()` for now, and we can loop at interrupts later.

> Unsure. Maybe spam the uart and check LQ?
Yep. good plan. Can you tackle this task?

> The current use cases I can think of are, changing mode, tx power, and vtx channel.
Lets start with the TX operations first, and think about how to get MSP over the air longer term.

JyeSmith, 2020-04-03:
Yep I will do the spam testing once you have a branch up.

Agreed, start with easy local tx changes.  We can worry about OTA once we have encapsulated crsf working.

JyeSmith, 2020-04-06:
Proposed [MSP V2](https://github.com/iNavFlight/inav/wiki/MSP-V2)

uint16 Function - ['E', 'x'] (e.g. Dec = 17784, Hex = [0x45, 0x78], Bytes = [69, 120])

**ExLRS function (uint8) and payload**

- Mode - 0x00, uint8 mode number 0, 1, 2, 3, 4
- Tx power - 0x01, uint16 mW
- TLM rate - 0x02, uint8  0, 2, 4, 8, 16, 32, 64, 128

**E.g. MSP V2 for setting ExLRS Tx power to 1W.**

[$, X, <, flag, function, function, payload size, payload size, ExLRS function, Tx power, Tx power, CRC]
[$, X, <, 0x00, E, x, 0x03, 0x00, 0x01, 0x03, 0xE8, CRC]

closed 2020-05-10 by wvarty