ExpressLRS/ExpressLRS #87: More cleanup in ESP32uartTask

JBKingdon, 2020-05-17:
I noticed a couple of opportunities for improvements in ESP32uartTask and left some comments (didn't want to pollute the PR I was working on with unrelated changes)

AlessandroAU, 2020-05-20:
Yep I saw those, we can implement them. Same here as I just merged in something I knew was working when we tried to fix those crashes.

AlessandroAU, 2020-05-20:
Where are the comments? I saw them in a PR but I'm can't find them right now.

JBKingdon, 2020-05-20:
Cool. I should update https://github.com/AlessandroAU/ExpressLRS/pull/86 to remove the comments and fix any merge conflict

AlessandroAU, 2020-05-20:
Oh whoops sorry I didn't mean to break your PR! I will test the PR on r9m and fix the conflict tomorrow :)

JBKingdon, 2020-06-02:
I think we covered this, closing.

closed 2020-06-02 by JBKingdon