ExpressLRS/ExpressLRS #88: Change rx upload speed for Linux compatibility?

JBKingdon, 2020-05-17:
The current upload speed for the rx doesn't work on Linux and it catches me out everytime. Would it be ok if we changed it to 460800?

AlessandroAU, 2020-05-20:
For the pass-through programming? You can't change it because the RX is already listening at 420000baud for the special characters to reboot into bootloader.

JamesKingdon, 2020-05-20:
No, this is for the initial direct flash. For some reason the speed we currently have doesn't work on Linux. Because I only do the initial flash rarely I always forget to change the upload speed and assume that I messed up the build of the rx. It would be likely to confuse someone new to the project as well :)

JBKingdon, 2020-05-20:
No, this is for the initial direct flash. For some reason the speed we currently have doesn't work on Linux. Because I only do the initial flash rarely I always forget to change the upload speed and assume that I messed up the build of the rx. It would be likely to confuse someone new to the project as well :)

AlessandroAU, 2020-05-20:
Ah right, does this happen for the ESP32 boards too?

JBKingdon, 2020-05-20:
No, the esp32 boards are fine.

AlessandroAU, 2020-05-21:
Ah ok, Are you using long hookup cables to connect the UART adaptor? I've seen that issue with long cables, I try to keep mine <10cm.

JBKingdon, 2020-05-21:
Interesting idea. These are actually pretty short. I should do some more testing and see if I can figure out what the underlying issue is.

cruwaller, 2020-05-21:
I haven't have any issue using Debian.
Which distro you are using?

JBKingdon, 2020-05-21:
Mint. From the sound of it, it's most likely a limitation of the particular uart adapter I'm using.

wvarty, 2020-06-21:
Missed this issue thread. FYI @JBKingdon uploading doesnt work reliably at 921600 either, and everytime I checkout a fresh branch, I need to manually set the rate to 115.2k before I flash, otherwise it fails.

AlessandroAU, 2020-08-31:
You guys still having a problem with 921600 ?

JBKingdon, 2020-08-31:
I always use 460800. The original problem reported was because the default came as a nonstandard speed that my Linux distro won't accept, but looks like they've been changed now. A quick test suggests 921600 is fine on my esp32 devkit. I don't have an rx handy to test on.

wvarty, 2020-08-31:
Uploading fails 99% of the time on 921600. I have to change to 115200 every time. It’s habit now so no big deal.

JBKingdon, 2020-08-31:
But if it fails for you it will fail for a new user who won't know what's going on.

greenbigfrog, 2021-01-13:
I haven't had any issues ever anywhere on DIY and R9 stuff.
*I use arch btw*

MonroeKA, 2021-01-15:
I use ubuntu as well no issues

brandonrc, 2021-05-09:
@JBKingdon is this still an issue? 

Considering we have a Configurator now I will close this issue and if we see flashing issue we can open up the issue on that side. 

@jurgelenas do you think this is something we can add to the configurator? I think we talked about this before on the discord.

closed 2021-05-09 by brandonrc