ExpressLRS/ExpressLRS #95: Is this open source? If so, it's missing a license.

platisd, 2020-05-29:
Cool project, but is this open source? I can look at the code but not much more than that.

If it's intended to be open source, could you please [add a license](https://choosealicense.com/)?

AlessandroAU, 2020-06-02:
License still not decided, but it will be open.
Ideas @JyeSmith @wvarty @JBKingdon @cruwaller @dabit20 ?

platisd, 2020-06-03:
Interesting... I look forward to you choosing a license. Aside of this, I am doing some small research so I have interest on the the following topic: Do you mind elaborating on how come you did not select a license to begin with, especially considering you are trying to create a *community* around the project?

At its current state, if anyone tries to do anything else to the project other just reading it, they are literally violating your copyright and could -in theory- be liable to legal action.

JyeSmith, 2020-06-03:
Hey @platisd 

I literally don't think anyone cares licensing.

This is a fun hobby project and we would be more than excited to see someone pick it up and further the software development, or release hardware commercially.

platisd, 2020-06-03:
Thanks for your response. It is particularly interesting. I'd suggest a permissive license in that case, such as the MIT.

May I get in touch with you again about this via your Discord at some point in the future?

AlessandroAU, 2020-06-03:
I would never be interested in litigation regarding this small hobby project, I assume none of the other devs are either. We have a handful of FPV pilots who are happy just work on something for fun. Time spent discussing a license is better spent testing firmware and flying with friends. 

I am not a lawyer nor a software developer by trade, this is all done out of passion to learn and explore. I'm not here to be picky about IP. For now, I promise not to sue you if you download the code. When the other developers provide their input we can define the official license.

platisd, 2020-06-03:
Haha, no worries, I didn't suspect you would. Your repository came into my attention after discussing with a Facebook friend and as an open source enthusiast (who is coincidentally running a research on the topic of repositories without a license) I thought it would be great if there was a FOSS license accompanying the otherwise great work you're doing here. :smile_cat:

JBKingdon, 2020-06-03:
I have no concerns or objections about adding a license. I haven't looked deeply into the issue, but It looks like either gpl or mit would be fine.

gretel, 2020-07-22:
`mit` be great.

maybenikhil, 2020-07-22:
MIT is the most common open-source license

Dope-johnny, 2020-09-16:
I think the license should allow the distribution of hardware kits (PCB design). That boost in availability could attract more developers and testers.

AlessandroAU, 2020-09-16:
What do we think about GPL3?

JBKingdon, 2020-09-16:
It's good, but it might make life more difficult for a manufacturer to pick up the code and sell it with hardware, iff they actually cared about the license and they wanted regulatory approval for the hardware. MIT would be better in that respect if we wanted to encourage uptake since it has few restrictions other than acknowledging the authors and keeping the license intact.

JBKingdon, 2020-09-16:
We might want to look at https://ohwr.org/cernohl for the hardware designs

greenbigfrog, 2020-11-28:
I'd like to bump this.
Having no license isn't really a viable option :wink:

JyeSmith, 2020-11-28:
@greenbigfrog why?

platisd, 2020-11-28:
I'll just jump in again. 

> @greenbigfrog why?

Potentially, any third party making a fork of this repository to submit a contribution is breaking the law. 😜
Same goes for anyone who is _using_, in almost any way, the resources found in this repository. So I guess going on without a license not being _viable_, is a good way to put it. :)

greenbigfrog, 2020-11-28:
Why? Basically what @platisd said
Additionally, agreeing on a license down the line will be even harder.
Choosing a license now can help avoid any potential drama down the line.

gretel, 2020-11-29:
a license will not be applicable backwards in time. if the code is used as is now it will be hard to sue the possibly commercial..military users ( `eachine` ... `raytheon` ) to make money..weapons of it eventually. even without having to mention the original authors. i still like the pirateish approach for the arrrrrr reason.

wucke13, 2020-12-26:
I'd propose the MIT OR APACHE 2.0 combination which is very popular in the Rust community. [The reasoning](https://prev.rust-lang.org/en-US/faq.html#why-a-dual-mit-asl2-license) is simple:

> The Apache license includes important protection against patent aggression, but it is not compatible with the GPL, version 2. To avoid problems using Rust with GPL2, it is alternately MIT licensed.

kenthinson, 2021-01-02:
I will add my two cents about license. If you don’t add one you are blocking any western company from making hardware for the project. Example RaceDayQuads. However Chinese companies will not respect the fact that there is now license and will start selling product as soon as they see a viable market. So if you would like to provide a even footing a permissive license like MIT would be good.

greenbigfrog, 2021-01-13:
Biweekly reminder there still is no LICENSE

jurgelenas, 2021-01-14:
A little summary of licenses used in our hobby space:

| Project  | License |
| ------------- | ------------- |
| [OpenTX](https://github.com/opentx/opentx)  | [GPLv2](https://github.com/opentx/opentx/blob/2.3/GPLv2.txt)  |
| [Betaflight](https://github.com/betaflight/betaflight)  | [GPLv3](https://github.com/betaflight/betaflight/blob/master/LICENSE) |
| [EmuFlight](https://github.com/emuflight/EmuFlight) | [GPLv3](https://github.com/emuflight/EmuFlight/blob/master/LICENSE) |
| [INAV](https://github.com/iNavFlight/inav) | [GPLv3](https://github.com/iNavFlight/inav/blob/master/LICENSE) |
| [ArduPilot](https://github.com/ArduPilot/ardupilot) | [GPLv3](https://github.com/ArduPilot/ardupilot/blob/master/COPYING.txt) |

There is a nice article why ArduPilot picked GPLv3: https://ardupilot.org/dev/docs/license-gplv3.html

jurgelenas, 2021-01-14:
Audit of libraries / code used and their licenses on master branch:

| File                                                                                                                                                                                                      | License                                | Compatible with GPLv3 | Compatibility check reference URL                                                                                                                              |
|-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|----------------------------------------|-----------------------|----------------------------------------------------------------------------------------------------------------------------------------------------------------|
| [STL/LICENSE.md](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/STL/LICENSE.md)                                                                                  | CC BY-SA 4.0                           | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#ccbysa)                                                                                               |
| [src/bootloader/src/Src/it.h](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/bootloader/src/Src/it.h)                                                       | BSD 3-Clause license                   | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#ModifiedBSD)                                                                                          |
| [src/variants/GHOST_ATTO/variant.cpp](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/variants/GHOST_ATTO/variant.cpp)                                       | LGPL 2.1 or later LGPL version         | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#LGPLv2.1)                                                                                             |
| [src/bootloader/src/Src/main.h](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/bootloader/src/Src/main.h)                                                    | BSD 3-Clause license                   | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#ModifiedBSD)                                                                                          |
| [src/variants/L432K/variant.h](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/variants/L432K/variant.h)                                                      | LGPL 2.1 or later LGPL version         | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#LGPLv2.1)                                                                                             |
| [src/variants/GHOST_ATTO/variant.h](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/variants/GHOST_ATTO/variant.h)                                           | LGPL 2.1 or later LGPL version         | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#LGPLv2.1)                                                                                             |
| [src/variants/r9mx/variant.h](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/variants/r9mx/variant.h)                                                       | BSD 3-Clause license                   | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#ModifiedBSD)                                                                                          |
| [src/lib/SX1280Driver/SX1280_hal.h](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/lib/SX1280Driver/SX1280_hal.h)                                           | Revised BSD License                    | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#ModifiedBSD)                                                                                          |
| [src/variants/R9MM/variant.h](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/variants/R9MM/variant.h)                                                       | LGPL 2.1 or later LGPL version         | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#LGPLv2.1)                                                                                             |
| [src/bootloader/src/Src/it.c](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/bootloader/src/Src/it.c)                                                       | BSD 3-Clause license                   | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#ModifiedBSD)                                                                                          |
[... cut ...]
| [src/python/xmodem.py](https://github.com/AlessandroAU/ExpressLRS/blob/92f06611c2893a7a11e539dc05bb6b31b3490fb9/src/python/xmodem.py)                                                                     | MIT                                    | :heavy_check_mark:                     | [Link](https://www.gnu.org/licenses/license-list.en.html#Expat)                                                                                                |

We can't really consider MIT license, because codebase contains more restrictive licenses like GPLv3. And you can't really wrap more restrictive license (GPLv3) with less restrictive one (MIT).

greenbigfrog, 2021-01-14:
the css CC3.0 incompatibly can very easily be solved. (not sure why some third party "bloat" was added in the first place :wink:)

jurgelenas, 2021-01-14:
I think [OpenTX binaries](https://github.com/AlessandroAU/ExpressLRS/tree/master/OpenTX) are "[GPLv2 or later](https://github.com/opentx/opentx/blob/0ee0c83d6f5ce5585be82539c0f11c3fe64def0f/GPLv2.txt#L299)" and that is compatible with proposed GPLv3.

Source: https://www.gnu.org/licenses/gpl-faq.html

AlessandroAU, 2021-01-17:
Thank you so much, for the digging guys, I guess from what I can see GPLv3 will be the most straightforward.

AlessandroAU, 2021-01-20:
Lets give a couple of week for anyone to voice their concerns, then we will officially label the project as GPLv3.

JBKingdon, 2021-01-20:
No objections

jurgelenas, 2021-03-03:
I think this issue can be closed now :)

platisd, 2021-03-03:
Cool! :+1:

closed 2021-03-03 by platisd