ExpressLRS/ExpressLRS #99: failsafe setting ? , failsafe mode

adrianmiriuta, 2020-06-12:
Is there a failsafe setting ?
I cannot get my FC into failsafe mode.

AlessandroAU, 2020-06-13:
Fail-safe is 'no pulses' as in it will stop transmitting packets to the FC. It can't be manually triggered yet.

closed 2020-07-26 by AlessandroAU

cptX, 2024-07-09:
@AlessandroAU I'm looking all over the internet to understand what will happen to the Receiver if it loses the TX signal. Here is the only reference I found stating that the receiver will not output any signal to the FC. I suggest to clarify this in the documentation. It is important to know that to ensure that the FC will detect the RX Loss and will execute the failsafe procedure. So are we sure the receiver doesn't output anything when losing signal? Betaflight and Inav configurators show the same signals when turning the TX off, so it's not clear what is happening.