ExpressLRS/ExpressLRS #100: R9M resistor mod question.

adrianmiriuta, 2020-06-14:
@AlessandroAU 
is the R9M resistor mod needed only for the Frsky X7 ?
or do you need it also for the X9D plus ?

JyeSmith, 2020-06-14:
Only the x7.

But ExpressLRS still works without the mod.

AlessandroAU, 2020-06-15:
I think there has been some confusion.

The X9D will do 400k and does not allow setting to 115200 baud. (AFAIK)
The QX7 cannot do 400k without the inverter mod, but it allows you to switch between 115200/400000.

The R9M module cannot do 400k without resistor mod.
Therefore, to use R9M at 400k with X9D requires the resistor mod.

I recommend doing the mod for maximum compatibility,

JyeSmith, 2020-06-15:
@adrianmiriuta ah, Ive given you bad info here which has caused all the issues :(

Ive updated the wiki with some more info regarding buad and x7 x9 handsets.

closed 2020-06-15 by JyeSmith