ExpressLRS/ExpressLRS #101: wrong roll pitch value transmitted for 10ms

adrianmiriuta, 2020-06-14:
@AlessandroAU 

in a 10 minute flight i had three bad input signal values
in roll and pitch (always about 10 ms) 
my firmware is for R9M tx, R9MM rx, compiled with EU freq band, and 3 pos switches.

[image]
[image]
[image]
[image]
[image]

JyeSmith, 2020-06-14:
Hi @adrianmiriuta can you please upload the log file.

adrianmiriuta, 2020-06-15:
@JyeSmith 

Hi Jye ,
attached config and log file.

[R9M_twitches.zip](https://github.com/AlessandroAU/ExpressLRS/files/4778439/R9M_twitches.zip)

JyeSmith, 2020-06-15:
Thanks @adrianmiriuta 

This is CF?  What debug was used?

adrianmiriuta, 2020-06-15:
@JyeSmith

debug_mode = ANGLERATE

you can see the settings in the diff file.

JyeSmith, 2020-06-15:
This is really weird and not something I have seen or felt in flight.  If it was spurious data I would expect it to affect all channels.  It is unlikely bad data from ExLRS since every packet has a CRC.

What gimbals are you using on the handset?  I know that I have had issues with slow response on hall gimbals caused by a bad connection.  But this looks different again.

adrianmiriuta, 2020-06-15:
@JyeSmith

clarification
I had the resistor mod like described in the rcgroups

[image]

I removed it because I got "Telemetry lost from my Taranis X9D plus"
and you said it is not intended for the X9D
and I got these twitches !
on my first flights with the resistor I didn't had them !

I have the  hall gimbals for  the Taranis.

JyeSmith, 2020-06-15:
Just checking, did you change the baud rate back to 115200?

Are you in the discord chat?  Might be easier to trouble shoot in there.

adrianmiriuta, 2020-06-15:
@JyeSmith

1. no (I use the TBS release of OPENTX  with CRSFshot fir X9D plus)
 I didn't find where to change serial speed!

2. no

adrianmiriuta, 2020-06-15:
@JyeSmith

I put the pullup resistor back in place, it does seem not to happen any more.
I have to fly a few packs to make sure.

br. adrian

AlessandroAU, 2020-06-15:
I think there has been some confusion. 

The X9D will do 400k and does not allow setting to 115200 baud. (AFAIK)
The QX7 cannot do 400k without the inverter mod, but it allows you to switch between 115200/400000.

The R9M module cannot do 400k without resistor mod.
Therefore, to use R9M at 400k with X9D requires the resistor mod

adrianmiriuta, 2020-06-15:
@AlessandroAU 
Thx.

closed 2020-06-15 by JyeSmith