ExpressLRS/ExpressLRS #146: Add ranging for SX1280

JBKingdon, 2020-08-12:
The SX1280 has a neat capability to do time of flight measurement to estimate distance between the tx and rx. This could be really useful for locating a fail-safed quad in tall vegetation for example.

maybenikhil, 2020-08-14:
https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/44000000MDiH/OF02Lve2RzM6pUw9gNgSJXbDNaQJ_NtQ555rLzY3UvY

https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/44000000MDcY/ZsmAVCVenZkc0lUrr3RuxWSfdFxY2Tj_msk4N9DAhBo

brandonrc, 2022-01-22:
@JBKingdon I know this is an old issue but I think one of the down sides is the reduced range when you go into that mode. 

Another down side is if a company uses sx1281 chips that do not support this functionality. 

There was an interesting idea about locating based on time it takes to get a telemetry packet. Not sure how accurate that would be. https://github.com/ExpressLRS/ExpressLRS/issues/892