ExpressLRS/ExpressLRS #149: R9MM + MINI Not syncing on first boot. starting in bootloader

l2smith2, 2020-08-14:
latest master branch flashed. 14/8/20
R9 RX's start directly into bootloader mode on first power up. power cycling they sync with radio.

cruwaller, 2020-11-09:
Sorry for late reply!
R9MM bootloader is updated (PR #206 ) lately and maybe you can try to update latest version.

AlessandroAU, 2020-11-09:
@l2smith2 did you have a chance to test the fix?

l2smith2, 2020-11-09:
I haven't tested the new bootloader. It seems the problem is most prominent
on quads with a large capacitor

gretel, 2020-11-11:
noticed the panasonic caps i use can take a couple of seconds to loose charge - which i have to wait in case.

l2smith2, 2020-11-11:
It's fine once the caps are charged. Just the initial power up causes the
problem

gretel, 2020-11-11:
ah nevermind i was refering to the time it takes until it resets properly..

gretel, 2020-11-15:
seems robust now - have been flashing the recent bootloaders.

AlessandroAU, 2020-11-19:
Thanks for checking @gretel 👍

closed 2020-11-19 by AlessandroAU