ExpressLRS/ExpressLRS #151: EBYTE E32 module support

axife, 2020-08-16:
These modules available on AliExpress for $5-10
They are based on SX1276 or SX1278 and STM8L0

QCZEK already did a close source firmware to make LRS from these modules.

JyeSmith, 2020-08-27:
Im not sure the stm8 will be powerful enough and at only 16MHz.  @peho-nordic has done some work with the STM32L071 (32MHz) and may be able to comment further.

JBKingdon, 2020-08-31:
Hi Jye - where did you find info on the cpu in the e32? I couldn't see anything about it in Ebyte's doc

axife, 2020-08-31:
QCZEK did LRS based on eByte. It is close source.
http://qczek.beyondrc.com/qczek-lrs-433mhz-1w-lora-rc-link/
> And finally i found E32-TTL-1W serial rf link based on LORA SX1278 spread
spectrum RF chip, and stm8L MCU, which would be re flashed by my firmware.

There are more references but I can't find right now. Possibly exact STM8L0
model is shown in YouTube video when someone flashes eByte

AlessandroAU, 2020-08-31:
It may be possible, but that chip is tiny, slow and doesn't have much flash space.

JyeSmith, 2020-08-31:
@JBKingdon the user manual states STM8L MCU.  Its either not going to work, or the amount of time to make it function isnt worth it.  These are used on VTxs and have bugger all space.

cruwaller, 2020-09-01:
I think it would be possible to get that working but requires lot of optimizations and simplifications.
And I guess you cannot get full benefit of ELRS to working with so small MCU.

AlessandroAU, 2020-11-09:
I think STM8 would be to limited to support the current features

closed 2020-11-09 by AlessandroAU