ExpressLRS/ExpressLRS #152: Stuck in bootlop after flashing TTGO V2.0 LoRa module

stylesuxx, 2020-08-17:
I seem to be running into a problem after flashing firmware to my TTGO V2.0 module from VSCode.

I hit "Upload and Monitor", building and upload works fine, but then I seem to be stuck in a boot loop:

```
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/DIY_900_TX_TTGO_V2_SX127x_via_UART/firmware.bin
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC: d8:a0:1d:49:e4:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10319...

Writing at 0x00001000... (100 %)
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.1 seconds (effective 937.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 7955.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 32774.3 kbit/s)...
Hash of data verified.
Compressed 264288 bytes to 135121...

Writing at 0x00010000... (11 %)
Writing at 0x00014000... (22 %)
Writing at 0x00018000... (33 %)
Writing at 0x0001c000... (44 %)
Writing at 0x00020000... (55 %)
Writing at 0x00024000... (66 %)
Writing at 0x00028000... (77 %)
Writing at 0x0002c000... (88 %)
Writing at 0x00030000... (100 %)
Wrote 264288 bytes (135121 compressed) at 0x00010000 in 2.3 seconds (effective 926.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
============================================================ [SUCCESS] Took 7.14 seconds ============================================================
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d13ec: ffffffff ffffffff ffffffff
Core 1 register dump:
PC      : 0x400d13f0  PS      : 0x00060130  A0      : 0x800d3602  A1      : 0x3ffb1f80  
A2      : 0x3ffc07e0  A3      : 0xffffff80  A4      : 0x00004e20  A5      : 0x80000020  
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x800d149a  A9      : 0x3ffb1f40  
A10     : 0xffffffff  A11     : 0x00061a80  A12     : 0x0800001c  A13     : 0x00000010  
A14     : 0x00000011  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d13f0:0x3ffb1f80 0x400d35ff:0x3ffb1fb0 0x4008b525:0x3ffb1fd0

Rebooting...
```
The same Exception with the same register dump then happens over and over again.

I am only connected via USB cable and have not yet hooked up BEC nor Radio - just in case that matters.

JBKingdon, 2020-08-17:
Problem was caused by serial2

 In user_defines, comment out -DUSE_UART2 :frowning: 
https://github.com/AlessandroAU/ExpressLRS/blob/576be147963c2183fde1c757a0ac3ef60ca90af4/src/user_defines.txt#L69

stylesuxx, 2020-08-17:
That solved it for me, thank you.

closed 2020-08-27 by JyeSmith