ExpressLRS/ExpressLRS #193: ESP32 backpack compatibility with other transceivers?

pablogranolabar, 2020-10-11:
Hi again,

I am wondering if this module (https://www.tindie.com/products/nicerf/lora1262f30-15w-sx1262-868mhz915mhz-rf-module/) could be used with your ESP32 back pack implementation? Could this module with the ESP32 backpack be a fully supported RX?

maybenikhil, 2020-10-25:
Currently the SX126x isn't supported as far as I know. The ESP backpack isn't an ESP32, it's an ESP8255, I would recommend going to the PCB tab here and looking at supported RF modules, etc.

AlessandroAU, 2020-11-09:
The backpack is mainly a developer debug tool, not really intended or ready for the average user yet. Also, you have linked a SX1262 module which is not 100% compatible with sx1276 current used in our designs and in the R9

closed 2020-12-07 by AlessandroAU