ExpressLRS/ExpressLRS #201: Weak telemetry link

JFly02, 2020-10-29:
I just finished setting up ExpressLRS for my R9M and R9mm (I love the projekt ;-)) but somehow the telemetry is breaking up as close as 2m from the rx and gets lost completely within maybe 7m. Any suggestions on how to fix it?
https://youtu.be/Dybz_BAp1HI

maybenikhil, 2020-10-29:
What telem ratio are you running?

JFly02, 2020-10-29:
1:64

AlessandroAU, 2020-10-30:
Is this at 200hz? Scan for sensors and show the LQ/RSSI so we can see how those look.

comsoft1908, 2020-10-30:
> I just finished setting up ExpressLRS for my R9M and R9mm (I love the projekt ;-)) but somehow the telemetry is breaking up as close as 2m from the rx and gets lost completely within maybe 7m. Any suggestions on how to fix it?
> https://youtu.be/Dybz_BAp1HI

Hope its okay to comment here though I'm not a developer. I have experienced similar thing with R9m TX and R9mm RX (EU 868 settings), but have maybe found a reason for this: If you set it to 50hz and leave the telemetry ratio at 1:64 the transmitted telemetry values are so "infrequent" that the QX7 seems to think that the connection is lost. If you change the ratio to 1:32 or lower the issue is fixed. However, if I am at 200hz I do not have that issue at all (well, I have not tested to set the telemetry ratio to something crazy low, but 1:64 definitively works fine for me at 200hz).

JFly02, 2020-10-30:
> Is this at 200hz? Scan for sensors and show the LQ/RSSI so we can see how those look.

https://youtu.be/zYH0QsT5YJY

JyeSmith, 2020-10-30:
That's pretty bad.  Leave it on the default 50mW, because it should be near perfect on that.  No need to up the power for the trouble shooting.

Do you have another antenna you can test?  Even give yours a wriggle and see if it gets better.  It could be broken internally.

JFly02, 2020-10-30:
Thanks, it actually was the receiver antenna, to dumb I didn't test that right away.

JyeSmith, 2020-10-30:
Awesome you solved it.

AlessandroAU, 2020-10-31:
@JFly02 would be good if you could also compare RSSIdbm at say 100m to what values we get. Even with a really bad antenna I wouldn't expect you to loose TLM so soon. It's possible you have a bad TX antenna or something else entirely.

JFly02, 2020-10-31:
With the new antenna I have no connection issues

AlessandroAU, 2020-10-31:
Awesome,

For reference you can see my RSSIdbm/LQ values around a typical track with approx. ~50m dist here https://www.youtube.com/watch?v=vublK1plq0s to compare.

closed 2020-10-31 by AlessandroAU