ExpressLRS/ExpressLRS #212: Use Esp8285 qfn32 instead of ESP-01 on Nano RX

rodrigosclosa, 2020-11-11:
It would be great if you could further reduce the size of the RX, using only the ESP8285 chip and the base components for it to work directly on the RX board ...

I have already used it in some personal projects and it is very easy to make this implementation ... What do you think?

JyeSmith, 2020-11-11:
Not a bad idea and is another option for diyers.  Wouldn't be hard to route if you went to a 4 layer PCB.

But I think the current design might be at the limit for a lot of us, and the next step is to just buy a ghost Rx.

JyeSmith, 2020-11-11:
FYI I'm dumb enough to attempt building it 😅

AlessandroAU, 2020-11-11:
if you want to design and contribute it please go ahead :) it's something I want to try

rodrigosclosa, 2020-11-11:
Great! I’ll do it forked from design in EasyEDA...

rodrigosclosa, 2020-11-11:
I've made the circuit with ESP8285 QFN32 package:

[image]

The PCB I'm still planning...

JyeSmith, 2020-11-11:
Wow, I never look at the ESP8285 that closely and didnt realise just how few components are actually needed.  This is awesome!

Is DTR and RST really needed.  I wonldnt replace it with a button, but just some tiny jumper pads for the initial flash.  You can see what I added on the the nano v1.1 PCB.

And depending on PCB room, a PCB wifi antenna in place of the ceramic.  It doesnt need to be good, just enough to stop the esp from burning up.  With that said... I dont use a wifi antenna on the current nano design... all good so far :|

You should be able to make a pretty tiny rx.  Even using castellated holes for gnd/5v/tx/rx to make the footprint tiny.

[image]

rodrigosclosa, 2020-11-12:
Yes, it's a much simpler design...
I preferred to put the DTR and RST for easy recording, without having to
press the button at the right time, etc.. If you need to record the
firmware several times, just connect, have it recorded and that's it ...
And add simple components on the plate, that don't change much in size..

I looked for some antenna designs on the PCB, there are several on
EasyEDA... But usually they are big, they are over the size of the board
... I will test some to see which is better in size...
I haven't added castellated holes yet, but it's easy to change. The only
thing is that instead of 4 holes, you will need 6, two more to plug in the
recorder directly and record the ESP8285...
Looking here, the PCB is more or less 17x12mm like that...

But another option would be to use an STM32F103RCT6, which has 256Kb of
Flash and needs few components on the tbm board ... It would be an option
to ESP8285, with a lower cost and ease of welding... I think it is the same
series used in the R9MM and R9 mini...

*Rodrigo Sclosa* • Arquiteto de Software

rodrigo.sclosa@gmail.com
(19) 99292.6385

 facebook.com/rodrigo.sclosa

AlessandroAU, 2020-11-12:
You don't need to use the transistors to invert the DTR/RST lines you can simply set a special config in platformIO and connect them directly. I use this on my 900mhz 20x20mm board. 

Another cool idea, the esp8285 can use up to a 52M crystal, it would be cool to figure out how the sx1280 and esp8285 can share the same crystal

JyeSmith, 2020-11-12:
> Yes, it's a much simpler design... I preferred to put the DTR and RST for easy recording, without having to press the button at the right time, etc.. If you need to record the firmware several [...]

What Sandro said above.

Once the rx is buried in a quad they will be never used.  So I would keep any pads minimal and not let them take up space.  If you intend for this rx to be used by others they may also cause confusion since DTR and RST are not found on other receivers.

> But another option would be to use an STM32F103RCT6, which has 256Kb of Flash and needs few components on the tbm board ... It would be an option to ESP8285, with a lower cost and ease of welding... I think it is the same series used in the R9MM and R9 mini... 

An STM32 on the repo would be nice, particularly if a vendor would like to pick it up for manufacture.  But I think most users would just buy a ghost rx if wanting stm.

rodrigosclosa, 2020-11-13:
@AlessandroAU I'll see your layout... I never used like this before... I think it's not possible to use the same crystal... Looking here for Ghost RX, it has 2 crystals... But I'll research about if it's possible...

@JyeSmith I have Ghost here and I've been seeing what it uses... The RX is really small and uses the STM32F301K8 which has 64KB of Flash... Its layout is also very simple, but it has two crystals, one for the STM32 and another for SX1280...

About the pads, I will follow the line most of the RX, which has the standard connection and pads on the recording board, in case you need to re-record for example if a brick happens...

FireNWater, 2020-11-15:
This is awesome.  The only holdup I have now for making Nanos is getting the ESP-01F chips.  Digi-Key has plenty of the ESP8252's.    Will you be able to keep the Nano board 2-sided with this design?

rodrigosclosa, 2020-11-16:
@FireNWater Yes, I'll keep it the same way as today...

JBKingdon, 2020-11-16:
@rodrigosclosa If there is an extra pin available on the 8285 can you hook it up to a second dio? It makes the code cleaner if we have two interrupts to work with.

maybenikhil, 2022-01-22:
closing because marked as stale for over a month

closed 2022-01-22 by maybenikhil