ExpressLRS/ExpressLRS #218: Prevent rx/tx link when switch mode is different

JBKingdon, 2020-11-18:
Currently there is nothing to prevent a tx and rx compiled with different switch modes from connecting, and since the switch positions will be mis-interpreter this is potentially dangerous as it could lead to unintentional arming.

Not sure how to implement this with the available bits - is there any space in the sync packet to indicate the switch mode? If not, we could change the caeser on the crc based on the switch mode which would be enough to at least reduce the probability of connection to a very low level.

nangel0, 2020-11-18:
And if we implement the openTx model binding number?

Not sure how it works, but tbs implemented it recently.

AlessandroAU, 2020-11-19:
Yeah I agree, needed safety feature. I think there's room in the sync packet.

cruwaller, 2020-12-17:
this is fixed already so closing now.

closed 2020-12-17 by cruwaller