ExpressLRS/ExpressLRS #223: link issues

awright424, 2020-11-22:
I have a 2018 r9m, accst vesrsion, and a r9mm-ota access receiver. I have successfully flashed both and just went through again to make sure that everything is the same for userdefines when flashed. I do not know if this is an issue with the hardware I chose being incompatible but initially this morning after turning my radio on first and then turning the quad on II was able to get a good link for about 5 minutes before I started having the drops. I am sitting here typing this with my radio not far from the receiver and have no connection. I am not sure what is going on. I have a video showing some of the issues I had: https://www.youtube.com/watch?v=ZwWq6rrEEis&feature=youtu.be

AlessandroAU, 2020-11-22:
Something definitely wrong, what setting have you selected in LUA, are they default? Did you do the resistor mod?

JBKingdon, 2020-11-22:
Hi @awright424 
You might want to hop onto our Discord server if you haven't already. It's a bit chaotic but they're a friendly bunch and will do what they can to help. There's a lot of people that have recently setup R9 gear on there at the moment so the chances are that someone will have seen a similar problem.

https://discord.gg/dS6ReFY

awright424, 2020-11-22:
> Something definitely wrong, what setting have you selected in LUA, are they default? Did you do the resistor mod?
> 
> EDIT: Would be good to check the communication between Radio and TX is good, in the upper right of the LUA script display you should see some numbers, eg 0:200. What are yours?

the lua script doesnt seem to indicate good packet health at all. I have seen it show 1:0 which was weird.

AlessandroAU, 2020-11-22:
that is not good, did you do the resistor mod?

awright424, 2020-11-22:
> that is not good, did you do the resistor mod?

no resistor mod as of yet. I need to look and hope I have resistors laying around that will work otherwise its order and wait.

AlessandroAU, 2020-11-22:
Do you have values show up in the LUA script?

awright424, 2020-11-22:
> Do you have values show up in the LUA script?

[image]
[image]

AlessandroAU, 2020-11-22:
That looks fine, are you still getting dropouts? Can you scan for sensors and show telemetry?

awright424, 2020-11-22:
its gaining and losing telemetry after connecting with a solid link for 4 minutes. 

https://youtu.be/S5bVk5LVeTE

i have two 470ohm resistors so gonna try the resistor mod now

AlessandroAU, 2020-11-22:
Looks like a UART comms issue to me, the resistor mod should fix it. A single 470ohm resistor should be fine

awright424, 2020-11-22:
well i got the resistor soldered on then when attempting to put the case together broke the pad off the inverter. this module might be done for. ordered another of the 2018 versions for 10 bucks

AlessandroAU, 2020-11-23:
It might be recoverable, post a picture of which pad got damaged.

awright424, 2020-11-24:
[image]
you cacan see the pad on the inverter. soldering to the smd resistor was fine. board turns on but no communication with the radio.

AlessandroAU, 2020-11-24:
Is the tab of the transistor broken? It's possible to bridge to the resistor next to it, do you have a hot-air station?

awright424, 2020-11-24:
> Is the tab of the transistor broken? It's possible to bridge to the resistor next to it, do you have a hot-air station?

no I do not. you mean where r3 is silk screened? i can try with a regular iron but i doubt it will work. hoping to have my replacement come Wednesday

AlessandroAU, 2020-11-24:
If you're up to it, you can replace the transistor you broke with one of these I highlighted in green (we don't use them) 

[image]

awright424, 2020-11-24:
so i moved the resistor with j6. i then soldered the 3.3v connector to the arm of the resistor. is that correct?

AlessandroAU, 2020-11-24:
Top of transistor goes to top of R3 with the extra resistor between those two and 3.3v. Can you send pic?

awright424, 2020-11-24:
i see what i did wrong. i have no connection to r3.

[image]

awright424, 2020-11-24:
looks like its working now. i can get the lua script to show data. did a pull compiling and flashing the tx and rx to be safe

AlessandroAU, 2020-11-24:
Nice!

awright424, 2020-11-24:
[image]
[image]
[image]
thank you for the great work and excellent response time. you saved me from having to buy a tbs tracer and i didnt have to spend more money. i did buy another 2018 r9m but at 10 dollars its good to have a backup. thanks again for the help and the great project. not once have i lost signal or telemetry while testing so far

AlessandroAU, 2020-11-24:
Happy flying 👍

closed 2020-11-24 by awright424

JBKingdon, 2020-11-24:
Hey, well done! Glad you got it going.

arfk9919, 2023-03-03:
I want to ask if there is an alternative to the transistor that I marked is damaged
[image]