ExpressLRS/ExpressLRS #270: Help Identifying Park from 2018 R9 Mod

russphelan, 2020-12-29:
I wasn't sure where else to ask this, but when adding the 1000ohm resistor on my 2018 R9 unit, I broke a leg off of the 3-legged SO-23 package (the thing that isn't the capacitor). Do you know what this part is? I have found an SO-23 package with the same 'J6' marking; it says it's an NPN transistor, part number 'mms9014'. Not sure if this is what I want, but I'd like to buy a replacement part. Thanks.

AlessandroAU, 2020-12-30:
mms9014 should work, otherwise you can use this transistor here will is used to invert the signal going to the pins on the back of the module. (J6 here) 
[image]

russphelan, 2020-12-30:
I'm a bit inexperienced. Are you suggesting that I can remove the J6 you marked, use it in place of my damaged one, then swap a connection so the signal no longer needs to be inverted?

AlessandroAU, 2021-01-02:
J6 marked on the image is not used at all in ELRS, you can remove it and replace your broken transistor with it. The image is from the backpack mod which is another thing entirely so don't worry about it. I just used it to show the location of the component.

AlessandroAU, 2021-01-11:
Did you manage to fix it?

closed 2021-01-22 by AlessandroAU