ExpressLRS/ExpressLRS #276: 433 MHz support

wucke13, 2021-01-01:
Is it possible to support the 433 MHz band as well?

greenbigfrog, 2021-01-01:
Mind if I ask why you're interested in a 433 MHz link? It should be quite possible and is how eLRS had started (according to what I found by a search in the discord server (link in the wiki)).

433 won't allow as high packet rates, and so far no one has managed to outfly 900 MHz.

AlessandroAU, 2021-01-02:
433mhz is already one of the supported configurations with a sx1278 based module. Although I must warn no one has tested that config in quite some time.

wucke13, 2021-01-02:
I was stupid and bought the wrong lora32 :laughing: . Also, I'm very interested in a opensource lora based radio link which is reliable, still in active development and provides good telemetry support/transparent serial connection. I know the last two points haven't been priorities in ELRS so far, but my hope remains :smile: . Grasping from the projects activity, tooling and documentation I'm estimating that this is the most promising development of which I'm aware ATM.

wucke13, 2021-01-05:
@AlessandroAU would you mind confirming whether `DIY_900_TX_TTGO_V2_SX127x_via_UART` is the correct target for a 433MHz ttgo v2.0 board?

AlessandroAU, 2021-01-06:
This should work, just define the 433mhz reg domain and upload.

wucke13, 2021-01-06:
Well, that seems to have worked. One last question, can the ttgo v2.0 function as receiver too, for 443 MHz?

AlessandroAU, 2021-01-06:
Unfortunately no, ESP32 boards can only be used for TX, not for RX

wucke13, 2021-01-06:
What rx is compatible with 433 MHz then?

AlessandroAU, 2021-01-06:
Any of the DIY designs using a 433mhz sx1276 module (eg RFM98)

AlessandroAU, 2021-01-11:
Let me know if you have issues with 433mhz, I am closing this issue for now.

closed 2021-01-11 by AlessandroAU

padcom, 2021-08-16:
Just a thought... 433MHz radio link would be awesome for penetration at long range. I'd give it a shot...

cmanley, 2023-11-28:
@greenbigfrog The only advantage I can think of is that it probably interferes less with 1.2Ghz VRXs than that 868/900Mhz does.

zhoujiage770-collab, 2026-10-18:
哎