ExpressLRS/ExpressLRS #305: (native) Dynamic Power

greenbigfrog, 2021-01-11:
Would reduce power consumption as well as reduce contribution to RF noise floor.

("requested" by Fernandez#5615)

( #124 is not really related to this)

JyeSmith, 2021-01-12:
How would we like to do this?  Based on rssidbm might be good since we already have rf mode sensitivity hard coded.

If with 20dbm of the sensitivity increase the output power?  Plus add some hysteresis control.

https://github.com/AlessandroAU/ExpressLRS/blob/f9c54b0fbbf71d677083d490da60516bc28e4a08/src/src/common.cpp#L18

greenbigfrog, 2021-01-12:
> If with 20dbm of the sensitivity increase the output power?
Linearly?

JyeSmith, 2021-01-12:
Power output would just be bumped up to the next level. https://github.com/AlessandroAU/ExpressLRS/blob/f9c54b0fbbf71d677083d490da60516bc28e4a08/src/lib/POWERMGNT/POWERMGNT.h#L59

e.g. for 200Hz mode, if you drop below -92 then increase power.  If you go above -82 then decrease power.

evildronepilot, 2021-05-27:
Is it possible to implement the behaviour similar to CSRF?
First add new option in lua script called dinamic power as a tick box.

IF ticked TX transmission power will start at minimum and progress (Base on lq, linearly or in steps) but only up to the preset power value?

SunjunKim, 2021-07-07:
This implementation, by James, looks promising.

SunjunKim, 2021-07-08:
I've implemented an elementary dynamic power function on top of the RC9 release.  Please try if you want:
https://github.com/SunjunKim/ExpressLRS/tree/c88d67a429812664d464b36ace8e225926e04076

Disclaimer: It's kinda working but not tested thoroughly. Use at your own risk.

SunjunKim, 2021-07-09:
> I've implemented an elementary dynamic power function on top of the RC9 release. Please try if you want:
> https://github.com/SunjunKim/ExpressLRS/tree/feature/dynamic-tx-power
> 
> Disclaimer: It's kinda working but not tested thoroughly. Use at your own risk.

https://www.youtube.com/watch?v=Ah6h-QqM6Xs
A very first test of dynamic TX power. The feature was working great, and a terrifying flight into a fog 🤣 .

PR #724 (2021-07-28): Dynamic tx power feature
This resolves #305.

This Dynamic Power features less power consumption, reducing the risk of potential overheating of a TX module, and less congested RF environment. 
The feature relies heavily on telemetry data, so it needs telemetry to be turned on (and ideally more frequently) to work smoothly.

A configured power on the Lua script will work as "MaxPower" set on the TBS Crossfire/Tracer system -- the output power will not exceed the configured power and only be reduced from it.

The power adjustment works in two ways:
1. RSSI-based slow speed power change: from an  [... cut]
closed 2021-07-28 by JyeSmith