ExpressLRS/ExpressLRS #308: SiYI Rx compatability

Dropsleutel, 2021-01-12:
I did my best to trace the Siyi diversity Rx, it seems based on SX1280 and SiGe2431L.
Flashing is very easy with STlink tested and it connects. (But readout protected)
It would be very much appreciated if this Rx could be ported for use as ExLRS Tx module and as a PA/LNA long range diversity Rx

Warning, pinout not guaranteed, all is very small and very hard to trace.

SIYI 2.4ghz diversity Mini Receiver

STM32F373 
PA10 Tx 1 - bidirectional main output ?? to 5pin component
PF6	 Rx 1 - bidirectional main output ?? to 5pin component

PA2	Tx 2 - 1kseries, serial out mini pad
PA3	Rx 2 - 1kseries, serial out mini pad

Pb9	Tx3 - 1kseries, serial out mini pad
PB8	Rx3 - 1kseries, serial out mini pad

PB3 	NSS_CTS SX1280
PB4	MiSo_Tx	SX1280
PB5	MoSi_Rx	SX1280
PE9	Busy  	SX1280
PE8	DIO 1	SX1280
N.C.	DIO 2 	SX1280

PA8	Antenna Select 	SiGe 2431L (ANT1=0v ANT2=3v3)
PD8 CTX 			        SiGe 2431L (3V3=Tx mode)
N.C.	CSD	set to 3v3 	SiGi 2431L (Rx LNA mode)
N.C.	CPS	set to 3v3	SiGi 2431L (Rx LNA mode)

Dual Color LED RED/GREEN not traced. May be it can be traced by a small test software alternating some pins?

Dropsleutel, 2021-01-24:
[image]
[image]
[image]
[image]
[image]
[image]

CapnBry, 2021-03-17:
I'm currently working on this, thanks for working out most of the pin mappings! I built the variant for STM32F373CC and have code running on it, lighting up the LEDs and sending out serial data. Here's the rest of the pin mappings:
PB6 LED RED, active low
PB7 LED GREEN, active low
PB2 SX1280 nRST
PA0 connected to "UAV"? pad through 220k resistor
Weirdly, the hardware UART3 isn't on those pins (PB8/PB9) but you're right they are definitely connected there. Maybe they're doing a software serial or something on it.

I can reset the SX1280 and the BUSY line works, but I don't get any SPI data back from it. Will have to get it hooked up to the scope somehow to see if my variant is actually working correctly or maybe one of the pins is wrong.

CapnBry, 2021-03-18:
Got this working, just need to finish up work on it by making the bootloader and doing some cleanup. Pull request coming soon.

Dropsleutel, 2021-03-18:
Thanks man looking forward to test, but I only have SiYi rx modules, so i hop at some point the Rx could be used as tx.

CapnBry, 2021-04-29:
SIYI FM30 RX and TX are supported by ExpressLRS 1.0. RX as a TX is on the project board for future development.

closed 2021-04-29 by CapnBry

FRPProject, 2022-04-20:
Thanks for your great work on this module.

I flashed my FR-Mini with ELRS-TX V2.4. The configurator indicated that the installation was successful, the LED lights up and becomes solid green after powering up. **However, I have connectivity issue between the radio and the FR-Mini-TX**. I have reached out to [i6x-openTX devs](https://github.com/OpenI6X/opentx/discussions/108) and it seem everything is correct in terms of the physical connection and the radio.

I followed [this tutorial](https://www.youtube.com/watch?v=uYnaEBqN0Z4) to make a simple Arduino-based oscilloscope. Here is the signal output from the module. **Is this working as intended?**
[image]