ExpressLRS/ExpressLRS #309: LUA profiles?

raffxj, 2021-01-13:
guys

how about each rate have its own 'profile' saving?

like =
50Hz 250mW  telem off
100Hz 10mW 
200Hz 100mW 1:64 telem
etc etc

that sort of stuff

greenbigfrog, 2021-01-13:
Currently this can partially be achieved by editing the defaults in https://github.com/AlessandroAU/ExpressLRS/blob/master/src/src/common.cpp to your needs.

Having dedicated option to support profiles via Lua would still be nice a nice addition.

Ideally we could still support this sometime in the future, although most people will probably not switch their settings too often. I'm on 200hz 10mW 1:128 all the time. With the results we've been seeing recently, changing only the power level would be sufficient for most :smile:

jurgelenas, 2021-12-11:
I think this can be closed since it is set per RX already?

closed 2021-12-24 by maybenikhil