ExpressLRS/ExpressLRS #310: Support for Flysky FRM302

SUPANOOB-ZA, 2021-01-13:
Hi Devs

Is it possible to extend this to the Flysky FRM302 hardware?

Thanks in advance

greenbigfrog, 2021-01-13:
Yes, it should be able to run eLRS on FRM302 hardware, it's just that no one has bothered tracing the pins etc.
At their price point it's also not really competitive at all to the DIY 2.4 Tx module. The tx module is a rather simple build, only combining different modules.

Some people tried using the flysky Rx for eLRS, not sure what the state of that was. ( CC @JamesKingdon : IIRC they bought some at some point to play with)

SpaghettiHands, 2021-04-18:
I have a FRM302 Unit that I would LOVE to flash with eLRS! what needs to be mapped out?

closed 2022-02-03 by github-actions[bot]

SUPANOOB-ZA, 2022-02-23:
Hi Devs,

*PLEASE OPEN THIS ISSUE*

The FRM302 is a 2 watt TX using the Semtech SX1281 2.4GHz LoRa chip. Information below ...

(link 1: https://www.semtech.com/products/wireless-rf/lora-24ghz/sx1281) 

(link 2: https://media.digikey.com/pdf/Data%20Sheets/Semtech%20PDFs/SX1280-81_Rev3.2_Mar2020.pdf )

(link 3: http://www.multirotorguide.com/news/flysky-frm302-afhds3-long-range-system/ )

[image]

With the above information is it possible to create a target to flash firmware to this TX ?

Thanks
Regards

rajnish22-glitch, 2022-02-23:
Hi , it has the same chip but the Power amplifier is different if the devs are ready to work on it i can provide them the necessary details as the module has a fixed gain amplifier which has been discontinued (AWB7232) and it has an rf switch as well, which has a really small package but i think the working should be the same as the ones in the diversity rx or [CG2214M6][https://www.cel.com/pdf/datasheets/CG2214M6.pdf] i have these chips for my DIY 4W tx and both of these look quite the same ill check the pinout in the weekend, but before this i want to know if the devs are interested.

SUPANOOB-ZA, 2022-07-10:
Was this ever completed?

commanderguy3001, 2022-07-10:
would be interesting to know what microcontroller is on that frm302 receiver, as that really is the most deciding factor if you can flash elrs to it or not

SUPANOOB-ZA, 2022-07-10:
It seems it was done... To a point.. I'm - waiting for more info... [image]

https://www.facebook.com/groups/636441730280366/permalink/1008298993094636/

nightflasher, 2022-09-12:
well well well...flysky has this new radio "[Elysium](https://www.flysky-cn.com/el18-specifications)"  with AFHDS-3, this could be the next ExpressLRS only radio!?