ExpressLRS/ExpressLRS #312: Support for 5 in 1 Multi

harry1453, 2021-01-14:
Would it be possible to support the 5 in 1 multi module TX found on the Jumper T18? It only supports R9 at the moment and this would be a great upgrade.

AlessandroAU, 2021-01-17:
Yes, planned todo. Code is done more or less but I don't have a T18 for proper testing.

RudiRamaekers, 2021-01-27:
Hi Alessandro,
How can I find out if the 5in1-elrs is ready ?
Will it be in one multi protocol package so the 900Mhz ànd the 2.4Ghz is available ?
I'd like to borrow you my T18 for testing, but it has some nasty issues right now .
(interference glitches on 868Mhz, if I touch the antenna it stops flickering)
What country you live ?
Regards & stay safe

joevo2, 2021-02-21:
Exciting times ahead https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/pull/404 plus HappyModel is releasing their [implementation](http://www.multirotorguide.com/news/happymodel-expresslrs-system-es915tx-and-es915rx/) too for 900mhz. They seems to be working on a 2.4G version too.

kenthinson, 2021-02-26:
@AlessandroAU Do you need a JP5in1 for testing? I would donate to the cause.

iz3man, 2021-05-18:
Is there a way to fit the module to a Taranis QX7 as well? I wouldn't mind doing some soldering etc ...

padcom, 2021-05-24:
Any update on this? I just bought the Jumper T18 and would like to give the ELRS a shot on my long-range Z84....

harry1453, 2021-05-25:
@padcom I also have a T18. I'm going to have a look at this soon.

padcom, 2021-05-25:
@harry1453 If you need help testing I'll be more than willing :) I think repurposing existing hardware is a great way to make opensource software :)

harry1453, 2021-05-25:
@padcom Thanks! I'll let you know if/when I get around to taking a look at it.

iz3man, 2021-05-25:
Do you think it's worth buying it yet? I got an 4in1 and don't need more at the moment, but would invest in a 5in1 if it's quite sure that is may be supported in the future.

harry1453, 2021-05-31:
@iz3man I am definitely sure that this **may** be supported in the future. There are no hardware limitations. But I don't know for sure that it **will**. I generally don't like buying hardware based on future software promises, so if I were you I'd wait

iz3man, 2021-05-31:
@harry1453 Thanks! I'm in no rush either. It's just WHEN it's available I want it NOW. :) And lead times from China are long. But you're right. Better wait than have another piece of electronic waste in the ever growing box.

frizzle101101, 2021-07-02:
I can help with any integration edits and testing on this I have the hardware.

tr4x1s, 2021-07-12:
I'd be willing to help with testing too - never use the internal module on my T18 Pro anyway ^^

SiscoUnderWater, 2021-08-13:
Any news on this, i would be willing to test it :)

RudiRamaekers, 2021-08-13:
> @padcom Thanks! I'll let you know if/when I get around to taking a look at it.

Hi Harry,
Do you have any updates on the 5in1 ELRS project ?
There are many people around the globe waiting for it....
Regards,
Rudi

RudiRamaekers, 2021-08-13:
Hi Harry,
Do you have any updates on the 5in1 ELRS project ?
There are many people around the globe waiting for it....
Regards,
Rudi

frizzle101101, 2021-08-13:
Honestly would agree with you im looking for the code fir this at this
point if you have some pls share

harry1453, 2021-08-20:
@RudiRamaekers unfortunately due to recent personal circumstances I have not been able to make any progress. I'll be sure to update this thread if I do.

frizzle101101, 2021-08-20:
Whos up to the challenge?

dazzdoodlz, 2021-09-30:
Any updates on the elrs for jp5in1

sdellava, 2021-10-05:
In waiting for this nice add-on, I've a question: ELRS will (hopefully) be usable on the 2.4 GHz band only or on both band supported by the J5IN1 tx module?

joevo2, 2021-10-05:
> In waiting for this nice add-on, I've a question: ELRS will (hopefully) be usable on the 2.4 GHz band only or on both band supported by the J5IN1 tx module?

JP5in1 would only work with 900mhz ELRS AFAIK (when developed). Someone said it wouldn't work with 2.4G cause the existing modules in the market doesn't have the chip required. Unless they make a new multi module with the required 2.4g chip by ELRS.

sdellava, 2021-10-05:
> modules in the market doesn't have the chip required

Tnx. 900mhz band is much better than the 2.4 Ghz for long range but, ... does exist a 900MHz ELRS receiver?

sdellava, 2021-10-05:
> JP5in1 would only work with 900mhz ELRS AFAIK (when developed). Someone said it wouldn't work with 2.4G cause the existing modules in the market doesn't have the chip required. Unless they make a [...]

Nice. 900MHz band is better for long distance. Can't wait to try it

sdellava, 2021-10-19:
No news on this interesting topic?

RudiRamaekers, 2021-10-20:
Unfortunatly the masterbrain behind the ELRS-idea has decided to put no more energy in this JP 5in1 implementation.
I'm waiting since january , when Allessandro said : "Yes, planned todo. Code is done more or less"
By now I think we might close this thread as well.....
I'm also very dissapointed, just like many people waiting for this solution.

sdellava, 2021-10-20:
> So theres a few ppl interested here does anyone want to do any development? im willing to help but haven't had much time
> […](#)

I'm glad to help. I've some dev skills and I've a Jumper T-18 pro. 

I can for sure help in testing and provide feedback or fix suggestion.

padcom, 2021-12-10:
Said to see it go... maybe later,...

sdellava, 2021-12-10:
Not strickly related to this topic but maybe interesting: EdgeTx just added the support to CRSF on the internal Sport. So it is now possibile to desing a new 5in1 module that can easly support ELRS whitout need to deal with the MPM protocol. I hope OpenTx will shortly follow.

gnuton, 2022-12-24:
Hi,
I am very new to ELRS, but not new to the opensource scene.
I got a Jumper T18pro and I want to continue this effort.

I took the  @AlessandroAU  patch  and now it can be merged into Multiprotocol code.
The patch is here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/compare/master...gnuton:DIY-Multiprotocol-TX-Module:jumper-5-in-1-elrs

I have not tested this code yet, since i got my jumper a few days back and I do not have a receiver yet.

sdellava, 2022-12-24:
> Hi, I am very new to ELRS, but not new to the opensource scene. I got a Jumper T18pro and I want to continue this effort.
> 
> I took the @AlessandroAU patch and now it can be merged into Multiprotocol code. The patch is here: [pascallanger/DIY-Multiprotocol-TX-Module@master...gnuton:DIY-Multiprotocol-TX-Module:jumper-5-in-1-elrs](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/compare/master...gnuton:DIY-Multiprotocol-TX-Module:jumper-5-in-1-elrs)
> 
> I have not tested this code yet, since i got my jumper a few days back and I do not have a receiver yet.

Great job!

I've a T18 pro and I normaly use Frsky R9M mini (868) that can flashed with ELRS, so when ready count on me to test the new release with my setup

frizzle101101, 2022-12-24:
What version of elrs did you make compatible...its baked into the multi firmware here im guessing so will it be stuck at a certain version untill future development?

gnuton, 2022-12-25:
@frizzle101101 I have just took the original @AlessandroAU patch and I made it building and compabile with the latest Multiprotocol-TX-Module code. 
The ELRS Code in my patch is so based on an old ELRS snapshots from Jul 2020: you can browse the code here https://github.com/ExpressLRS/ExpressLRS/tree/dc6f4f21962fb04008bcb09c2a5379f3474e8bba

The entire ELRS part must be refreshed. The biggest issues I do see is that ELRS development goes really fast, so a new build of MP-TX-M is required which ELRS release and this would nicely work if I manage to import most of the code straight from this repo maybe using include files.. but again I have not checked ELRS so much yet.. very new still to this project and  I may need some guidance.

frizzle101101, 2022-12-25:
I would say following the betaflight spi releases would be the best so like what betaflight 4.4 will be doing support for elrs 3.0 compatibility because at this point
Bf are also all thinking about compatibility with baking in another projects code, so aligning with those versions used is most realistic in my opinion

gnuton, 2022-12-26:
@frizzle101101 sounds like a good idea.

Just for reference,  the code that we are talking about is pretty much here https://github.com/betaflight/betaflight/commits/00203a98e2807663883a7e2d922e9ff53abc9983/src/main/rx/expresslrs_common.h

BTW I do see an issue here, at this point Betaflight and MultiProtocol projects do require an "easy" way to integrate ELRS in other prjects, and maybe there will be more in the future.. Sounds like libelrs or something like this may be useful.

AlessandroAU, 2022-12-28:
Honestly, this is more a headache then it's worth, just get a cheap external module. This development and not planned and even discouraged. SPI in betaflight is already enough of a pain without also fragmenting this out. The POC code I did is so old it would need a complete rewrite, not to mention reimplementing the new modes, telemetry loan features etc.

There's also a the issue that the MPM modules don't support mixer sync (crsfshot) and are limited to around 140hz because of some old design decisions and needing to use software bitbanged serial on some radios. 

It's not worth the hassle. It would be better to just get a cheaper external module. This development is no longer planned and is actually discouraged. Betaflight's ELRS SPI is already difficult deal with, so adding this additional fragment will only make it more challenging. The proof of concept code I created is very outdated and would require a complete rewrite, not to mention the effort to implement new modes, telemetry loan features, etc. Additionally, the MPM modules do not support mixer sync (crsfshot) and are limited to around 140hz due to old design decisions and the need for software bitbanged serial on certain radios.

sdellava, 2022-12-28:
> It's not worth the hassle. It would be better to just get a cheaper external module. This development is no longer planned and is actually discouraged. Betaflight's ELRS SPI is already difficult [...]

Very agreeable remarks that nevertheless lead me to think that it would be very interesting to design an internal module, diviamo a V2 multiprotocol, that supports dual band ELRS.

Flying in FPV, and using rx In 868 (R9M) I never had failsafe: the video signal is definitely more critical, so it would be definitely unnecessary to use an external ELRS module.

The real reason for preferring ELRS is the greater penetration of walls and trees in 2.4 GHz

So my idea would be to design a 2in1 module, reusing Jumper's 868/900 module and replacing the 2.4 module with a native ELRS module.

It would not be a high-power module so as not to have to deal with heat dissipation, but despite this it would definitely perform better

Obviously the optimum would be a 2in1 both ELRS.

Any suggestions on how to approach this project?

AlessandroAU, 2022-12-29:
A few people have discussed this idea in the discord and James has even made some prototype hardware, I would suggest you join and start chatting :)

sdellava, 2022-12-29:
Perfect. I'm already there, but there is lot to read and I've missed this topic