ExpressLRS/ExpressLRS #313: 868MHz jumper R900 mini?

corrado-c, 2021-01-14:
Hello!
First of all: GREAT WORK!  THANK YOU SO MUCH!
I have a doubt about the R900 mini, with this FW it also works at 868?

i was looking for a hardware to make RX and it seems the cheapest.
THANKS!!!!!!

AlessandroAU, 2021-01-14:
It will work at 868mhz no problem 👍

closed 2021-01-14 by AlessandroAU

greenbigfrog, 2021-01-14:
Be aware though, that you need a STLink to flash the BL, as well as need to solder to the chip (at least that's the last information I have)  https://raw.githubusercontent.com/AlessandroAU/ExpressLRS/master/img/r900mini-rx/r900mini-side2-closeup.jpg

corrado-c, 2021-01-14:
thank you so much!  this is useful to me, I didn't notice, but it's fine the same, no problem!
I just have to figure out which pads to use to flash and which to connect to the FC