ExpressLRS/ExpressLRS #317: 900mhz DIY TX files

darwinfpv, 2021-01-16:
I can't find the pcb files for the 900mhz TX... Sorry I'm kinda new on github 😅 in the 900mhz pcb folder I only find the rx not the tx... Any help?

greenbigfrog, 2021-01-16:
That is, because there aren't really any 900 DIY TX designs, since TTGO as well as R9 transmitter stuff is available very cheaply. [Click here for supported 900 off the shelf hardware](https://github.com/AlessandroAU/ExpressLRS/wiki/Supported-Off-The-Shelf-Hardware#tx-1)
You are also more then welcome to join us on discord at http://discord.gg/dS6ReFY

AlessandroAU, 2021-01-17:
Not part of the official board yets but take a look here:

https://easyeda.com/cruwaller/exlrs-2-4-stmf103-tx
https://easyeda.com/cruwaller/expresslrs-tx-sx1280

trogalko, 2021-03-05:
I found the 868Mhz version here : https://oshwlab.com/enrico1036/expresslrs-tx-sx1280

benyi-dor, 2021-04-29:
So if i get one TTGO LoRa V1/V2 esp32 board, i need to flash it with ExpressLRS Configurator ?

Then wich pin do i need to connect to Sport on the esp32 module ?

benyi-dor, 2021-05-01:
Ok, I get my answer in the elrs discord, if needed : Have to plug S.port to pin 13.

brandonrc, 2021-05-09:
Closing this issue because @AlessandroAU and @greenbigfrog answered the OP.

closed 2021-05-09 by brandonrc