ExpressLRS/ExpressLRS #321: GPS telemetry

BatuP, 2021-01-18:
Are there any plans to add GPS telemetry to ExpressLRS? Additionally, what are some good resources to look at to start understanding the code base? I am very interested in learning how ExpressLRS works and how to potentially contribute to the project.

dyon-b, 2021-01-18:
For the telemetry part you should check out #204, It’s just waiting to be merged at this point.

jejakmedia, 2021-01-22:
i hope gps telemetry can read on Telemetry Viewer app in android

nangel0, 2021-04-17:
Advanced telemetry has been merged.
Can we close this @BatuP

ballistRC, 2021-04-25:
I was checking telemetry just now and while I noted GPS data, there wasn't any heading information so I dont think magnatometer  data is being forwarded.  This was using BF4.2.8 and the heading was changing when I rotated the Mag.

schugabe, 2021-04-28:
> I was checking telemetry just now and while I noted GPS data, there wasn't any heading information so I dont think magnatometer data is being forwarded. This was using BF4.2.8 and the heading was changing when I rotated the Mag.

The telemetry sensors are transmitted as received by the FC and betaflight mentions GPS heading in a comment. I think that the mag is not used for this field.

CapnBry, 2021-04-29:
Advanced Telemetry supports `CRSF_FRAMETYPE_GPS` which includes lat, long, ground speed, ground course, altitude, and num sats. I can confirm this is working in iNav (and should also work with betaflight since the code is the same) and displaying on opentx.

closed 2021-04-29 by CapnBry