ExpressLRS/ExpressLRS #322: Ghost TX Support

atr2600, 2021-01-19:
This is what I have so far. 
```
BUSY = PB15
DIO1 = PB2
MOSI = PA7
MISO = PA6
SCK = PA5
NSS = PA15
RST = PBO
LED = PB6
```

atr2600, 2021-01-19:
[image]

atr2600, 2021-01-19:
[image]

atr2600, 2021-01-19:
[image]

atr2600, 2021-01-19:
[image]

atr2600, 2021-01-19:
[image]

atr2600, 2021-01-19:
[image]

atr2600, 2021-01-19:
[image]

mavericm1, 2021-01-19:
[https://i.imgur.com/CZHvgLg.jpeg](url)

mavericm1, 2021-01-19:
https://i.imgur.com/51qPYOj.jpeg

mavericm1, 2021-01-19:
https://i.imgur.com/JeqDSib.jpeg

mavericm1, 2021-01-19:
https://i.imgur.com/X87XGgu.jpeg

mavericm1, 2021-01-19:
https://imgur.com/a/eSphNFu

atr2600, 2021-01-25:
[image]

howels, 2021-03-06:
Saw the ghost TX bootloader PR, what's the current status of ghost TX support?

AlessandroAU, 2021-04-03:
It's ready :)

closed 2021-04-03 by AlessandroAU

howels, 2021-04-03:
> It's ready :)

Tx lite or only the full size?