ExpressLRS/ExpressLRS #326: Frsky R9M Pro Lite,  Target

Dropsleutel, 2021-01-24:
I did trace and Map the Frsky R9M Lite Pro 1Watt compact Aluminium module:  
CPU STM 32F303 CCT6

Pb12 - Nss (Semtech 1278)
Pb13 - Sck (Semtech 1278)
Pb14 - MiSo (Semtech 1278)
PB15 - Mosi (Semtech 1278)
PA8  - Dio1 (Semtech 1278)
PA9  - Nreset (Semtech 1278)

PA4 - Vapc1 (SKY65111 33dbm Amplifier RF power control pin 1st-2nd Rf stage, Max 2.8V)
PA5 - Vapc2 (SKY65111 33dbm Amplifier RF power control pin 3rd RF stage, Max 2.8V)
n.c.- Vref = Vdd (SKY65111 33dbm Amplifier RF power control pin)

Pa6 - CTR (SKY13330 antenna switch 0Volt=Rf1-Rf3=Transmit 1.65-2.7Volt=Rf1-Rf2=Receive) 
n.c.- Enable = Vdd (SKY13330)    

PC14 - (AD8313 Rf Detector Voltage feedback from coupler circuit)

PA2 - Serial Pxx Rx (Tx/Rx, Pins could be swapped)
PA3 - serial Pxx Tx (Tx/Rx, Pins could be swapped)

PB3 - Led1 (color 1, don't know which color)
PB4 - Led2 (color 2, don't know which color)
PA14 - Led3 (color 3, don't know which color)

Note: 
Serial IO Pxx tx/Rx from module go via some serial buffer chips / level shifters, I do not have datasheet and do know the type of component marked C40R

I am happy to flash/test or assist in some more tracing

Dropsleutel, 2021-01-24:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Dropsleutel, 2021-01-24:
Indeed pinout post 1,  is untested, at your own risk.

quadrnk, 2021-01-29:
Happy to assist testing as well.

Jahnkeanater, 2021-02-07:
Here's what I was able to trace.

I'm not sure if PA8 is DIO0 or DIO1

PB3 is the red led
PB4 is the blue led
PA15? is the green led

I also managed to trace a few pins to the 26SU chip. I'm not sure if anyone ever figured out exactly what that is but I assumed it was an RF1126 RF switch.

[image]

I'm still unsure what these pins are doing but I assumed it has something to do with the SWR measurement.

#define GPIO_PIN_VRF1			PA7  // 26SU Sample RF1
#define GPIO_PIN_VRF2			PB1  // 26SU Sample RF2
#define GPIO_PIN_VRF2			PA0  // SWR? ADC1_IN1

Here are some pictures to show how I traced out these pins.

[image]
[image]

It's interesting that the Analog Devices chip "J 1A" on yours is not laser etched.
Looks a lot like an AD623 Instrumentation Amplifier.

[image]

Jahnkeanater, 2021-02-07:
I need to measure PA7 and PB1 to see when they are changing. I wonder if its just based on power output or if its constantly switching between the two.

Measuring PA0 at different RSSI/SWR levels is also on my to do list.

Is PC13 connected to anything?

Dropsleutel, 2021-02-07:
We are almost same findings,  What is PA0 connected to according you ?

Jahnkeanater, 2021-02-08:
Alright so here's what I found so far.
C40R TI SN74LVC2G240 Inverter
402WT ST M95160 2Kb EEPROM
J 1A Analog Devices AD8313  demodulating logarithmic amplifier
26SU RFMD RF1126 SWR RF Switch
STM32F303CCT6
SKY13330 Output RF Switch
SKY65111 PA
Semtech SX1276

#define GPIO_PIN_RFamp_APC1           PA4  //2.7V
#define GPIO_PIN_RFamp_APC2           PA5  //100mW@590mV, 200mW@870mV, 500mW@1.093V, 1W@1.493V
#define GPIO_PIN_RFswitch_CONTROL     PA6  // confirmed  //HIGH = RX, LOW = TX
#define GPIO_PIN_NSS            PB12 // confirmed
#define GPIO_PIN_DIO0           PA8  // confirmed
#define GPIO_PIN_MOSI           PB15
#define GPIO_PIN_MISO           PB14
#define GPIO_PIN_SCK            PB13
#define GPIO_PIN_RST            PA9  // NRESET
#define GPIO_PIN_SDA            PB7
#define GPIO_PIN_SCL            PB6
#define GPIO_PIN_RCSIGNAL_RX    PB11 // s.port inverter
#define GPIO_PIN_RCSIGNAL_TX    PB10 // s.port inverter
#define GPIO_PIN_LED_GREEN      PA15 // Green LED
#define GPIO_PIN_LED_RED        PB3  // Red LED
#define GPIO_PIN_LED_RED        PB4  // Blue LED
#define GPIO_PIN_DEBUG_RX    	PA3  // inverted UART JR
#define GPIO_PIN_DEBUG_TX    	PA2  // inverted UART JR
#define GPIO_PIN_VRF1			PA7  // 26SU Switch RF1
#define GPIO_PIN_VRF2			PB1  // 26SU Switch RF2
#define GPIO_PIN_SWR			 PA0  // SWR ADC1_IN1
//#define GPIO_PIN_EEPROM_DATA     PB8 //2Kb EEPROM ST M95160
//#define GPIO_PIN_EEPROM_CLOCK    PB9 //2Kb EEPROM ST M95160

[image]
[image]
[image]

VladimirP1, 2021-02-10:
Have you got S.port working? I have bricked my R9MLP, so really interested in getting ELRS working.Now neither the bootloader for R9M, nor your ELRS branch is able to talk with opentx.

VladimirP1, 2021-02-11:
@Jahnkeanater PB2 is OE for the S.port inverter

Jahnkeanater, 2021-02-12:
> 
> 
> @Jahnkeanater PB2 is OE for the S.port inverter

Is that 1OE or 2OE? I did verify PB2 but I forgot to document what pin it was on the Inverter.

Is this correct?

[image]

Jahnkeanater, 2021-02-14:
https://github.com/Jahnkeanater/ExpressLRS

@VladimirP1 @qdrk 

Try my fork and let me know what happens. No guarantees.

VladimirP1, 2021-02-14:
@Jahnkeanater I haven't tried flashing this particular commit, but seems you are missing RF switch control. And the other thing which is wrong is that BUFFER_OE is inverted, it means that OE will be active, when we do not want and we won't recieve any CRSF packets. I have got my R9MLP working, I fixed the OE problem, did someting like this `#define GPIO_PIN_RX_ENABLE      GPIO_PIN_RFswitch_CONTROL
` and also, I added code to power down the PA before switching to RX (I do not know if this is neccesary).

Dropsleutel, 2021-02-22:
Have you had some more time for testing ?

moyouyilansefengye, 2021-03-01:
Hi，guys.
A friend of mine wants to take part in the test. Could you tell me how to test the firmware?
He has flashed his r9m lite pro using VScode. However it wont be detected by radio. 
So we want to help testing the target:)

VladimirP1, 2021-03-01:
@Jahnkeanater Any plans for continuing fork on this? I have working code derived from your branch and I have already tested it for several flights. Should I make a PR into your fork?

@moyouyilansefengye The reason it is not detected by radio is that BUFFER_OE is inverted compared to normal R9M. Threre's one other problem with this current code: the RF switch is not switched - always in TX mode.

AlessandroAU, 2021-03-01:
If you have it working please make PR I have been lazy/not enough time for this. I am mainly flying 2.4ghz now :|

moyouyilansefengye, 2021-03-01:
> @Jahnkeanater Any plans for continuing fork on this? I have working code derived from your branch and I have already tested it for several flights. Should I make a PR into your fork? [...]

That's great！ 
Looking forward to your code for testing.

VladimirP1, 2021-03-01:
@moyouyilansefengye Try the **r9mlp** branch from here https://github.com/VladimirP1/ExpressLRS. It may be buggy, because that is not the exact code I am flying, I tried to make it less ugly) And I am too lazy to solder ST-LINK again to my R9M LP for now.

I have disabled the EEPROM for now, because the EEPROM chip used is an SPI one and will be hard to get working.

comsoft1908, 2021-03-01:
@VladimirP1 Did a quick bench test with your branch r9mlp connecting to a R9mini rx on latest develop branch from AlessandroAU (868_EU, fixed binding phrase). Looks very good, connects immediately, RSSI very good, LQ stable, error rate in ELRS LUA is 0:200 (perfect). 

Flashing only works through ST-Link (as expected), trying to flash .elrs file (after first initial flash with ST-Link) directly through OpenTx does not work (works for my R9M Lite that I also initially had to flash with ST-Link).

Is there a chance to get the powerlevels below 100mw added? I can see that @Jahnkeanater has added the power level definitons for 100, 250, 500 and 1000mw in POWERMNGMT.cpp already. I sthere anything I can do to help you with that or any other aspects of getting R9M Lite Pro fully supported?

comsoft1908, 2021-03-01:
[image]

For reference for anyone who wants to flash through ST-Link method here is a picture of the required connection: open the R9M Lite Pro, connect as seen in the picture (from left to right): 3.3V (red), SWDIO (white), SWCLK (green), GND (black)

moyouyilansefengye, 2021-03-02:
> @moyouyilansefengye Try the **r9mlp** branch from here https://github.com/VladimirP1/ExpressLRS. It may be buggy, because that is not the exact code I am flying, I tried to make it less ugly) And I [...]

OK，thanks. Will test later :)

VladimirP1, 2021-03-02:
The EEPROM chip turned out to be actually an I2C one, **ST M24C02**.  Fixed EEPROM pins and enabled it back. I rebased the code onto latest `develop` branch and I think the code is ready to merge, #366

comsoft1908, 2021-03-02:
> Is there a chance to get the powerlevels below 100mw added? I can see that @Jahnkeanater has added the power level definitons for 100, 250, 500 and 1000mw in POWERMNGMT.cpp already. 

I have reviewed the discussion back when the powerlevels were found and have, based on my understanding of this discussion, added the following powerlevels to POWERMNGMT.cpp:

    case PWR_10mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 293); //0-4095  234mV # deducted from 100-1000mW
        CurrentPower = PWR_10mW;
        break;
    case PWR_25mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 398); //0-4095  319mV # deducted from 100-1000mW
        CurrentPower = PWR_25mW;
        break;
    case PWR_50mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 542); //0-4095  433mV # deducted from 100-1000mW
        CurrentPower = PWR_50mW;
        break;

According to RSSI value this seems to be quite okay, but I have no power meter / antenna tester to be 100% certain.

These powerlevels were deducted from the powerlevels 100-1000mW and the following (mathematical) findings:

1.) doubling the power level results in ~ +1.73 times the mV value
2.) mW = mV * 0.8

Maybe someone with a antenna tester can confirm these settings?

digaus, 2021-03-07:
> [image]
> 
> For reference for anyone who wants to flash through ST-Link method here is a picture of the required connection: open the R9M Lite Pro, connect as seen in the picture (from left to right): 3.3V (red), SWDIO (white), SWCLK (green), GND (black)

Thanks for the info :) have you tested on 1000mw? How hot does the module get? Does everything work (lua etc). Which branch did you flash?

Jahnkeanater, 2021-03-07:
Sorry I have been busy moving into a new house the past couple weeks. The laptop charger is still MIA and I'm spending 50% of my time looking for things these days. The only reason I didn't use the default power levels was because the stock firmware doesn't have those outputs so I couldn't measure them to get DAC voltages. If someone has an RF power meter they can figure out what voltages are needed.

comsoft1908, 2021-03-08:
> Thanks for the info :) have you tested on 1000mw? How hot does the module get? Does everything work (lua etc). Which branch did you flash?

I have not yet tested on 1W, but I can already say that the module gets way warmer with ELRS than with R9 firmware (as expected due to ELRS higher duty cycle). 

LUA works (latest dev branch version). Just be aware that settings are not safed to module due to EEPROM is an SPI one which is not yet implemented in ELRS. So be sure to set the defaults in the code as you like or adjust that every time you go flying.

I flashed branch **_r9mlp_** from here https://github.com/VladimirP1/ExpressLRS including my power level adjustments as posted above (have meanwhile changed the values a bit since they seemed to be a bit too low - but that is purely measured based on RSSI in comparison to a R9M 2018 module and a R9M Lite module). These are my current settings for the power levels below 100mW (be aware: changed the default to 25mW):

```
#ifdef TARGET_R9M_LITE_PRO_TX
    Radio.SetOutputPower(0b0000);
    //Set DACs PA5 & PA4
    switch (Power)
    {
    case PWR_10mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 368); //0-4095  295mV # deducted from 100-1000mW
        CurrentPower = PWR_10mW;
        break;
    case PWR_25mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 425); //0-4095  340mV # deducted from 100-1000mW
        CurrentPower = PWR_25mW;
        break;
    case PWR_50mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 575); //0-4095  460mV # deducted from 100-1000mW
        CurrentPower = PWR_50mW;
        break;
    case PWR_100mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 787); //0-4095  590mV # adjusted, seemed to be too low
        CurrentPower = PWR_100mW;
        break;
    case PWR_250mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 1080); //0-4095 870mV this is actually 200mw
        CurrentPower = PWR_250mW;
        break;
    case PWR_500mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 1356); //0-4095 1.093V
        CurrentPower = PWR_500mW;
        break;
    case PWR_1000mW:
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 1853); //0-4095 1.493V
        CurrentPower = PWR_1000mW;
        break;
    default:
        CurrentPower = PWR_25mW;
        analogWrite(GPIO_PIN_RFamp_APC1, 3350); //0-4095 2.7V
        analogWrite(GPIO_PIN_RFamp_APC2, 398);  //0-4095 319mV
        break;
    }
#endif

```

digaus, 2021-03-08:
@comsoft1908 

@VladimirP1 mentioned he got the eeprom working since it is in fact a i2c chip:

https://github.com/ExpressLRS/ExpressLRS/issues/326#issuecomment-788787055

qdrk, 2021-03-12:
Just flashed my R9 LITE PRO and a R9MM receiver using this branch: https://github.com/VladimirP1/ExpressLRS/tree/r9mlp-rebase. Everything seems to work well.  Only glitch is after chaing the settings in LUA script, the link doesn't recover by itself, had to restart the radio to get link re-established (at least that's what's shown in the LQ OSD).

qdrk, 2021-03-12:
Great work folks!

molneo, 2021-03-16:
Flashed my R9MM-OTA and R9ML PRO with Vladimir's branch (https://github.com/VladimirP1/ExpressLRS/tree/r9mlp-rebase). Everything ok, except it cannot be compiled with DHYBRID_SWITCHES_8 which means that there are only 2 positions used for the 3 position switches. Could this feature be added in the next update? Great job everyone!

VladimirP1, 2021-03-16:
@molneo Seems like a problem with the configurator. If you select 'Manual mode' and set DHYBRID_SWITCHES_8, the firmware builds successfully.

VladimirP1, 2021-03-16:
@molneo My fork can be considered obsolete now, the PR #366 has been merged into the *develop* branch.

molneo, 2021-03-16:
@VladimirP1 thanks for the info, I'll look into it.

Dropsleutel, 2021-04-02:
Hi guys just to thank you for all your progress. I want to confirm I flashed with PIO via ST Link the R9M Lite Pro module and it went flawless!  After initial flashing via ST Link I can confirm, that you can flash from open Tx, by selecting the .elrs file.

So hope it inspires some more people to flash their R9M Lite Pro to Express LRS.
Thanks guys!

s2kaka, 2021-04-08:
> Hi guys just to thank you for all your progress. I want to confirm I flashed with PIO via ST Link the R9M Lite Pro module and it went flawless! After initial flashing via ST Link I can confirm, that it is only required one time only, you can than remobe ST link and future flash upgrades from open Tx, by selecting the .elrs file.
> 
> So I hope it inspires some more people to flash their R9M Lite Pro to Express LRS.
> Thanks guys!

Nice to heard that from you, I have a R9m LP too and looking for support from ELRS, 
How was your experience with it? Can I get telemetry like Frsky FW can do?

comsoft1908, 2021-04-08:
> Can I get telemetry as Frsky FW can do?

Telemetry (as far as what is supported by standard ELRS) works as with all other RX's. R9MLP just works without any issues, I used it for 3 weeks now as my primary RX. One thing I'd recommend is to add the lower power levels as suggested by me in this thread (these are not exact though, just rough estimations based on rssi values), otherwise you're stuck with 100mW (or more) which is not needed at all and makes the RX a bit warm.

s2kaka, 2021-04-08:
As i know basic telemetry is currently provided (VBAT, downlink/uplink LQ and downlink/uplink RSSI). 
But I saw some picture that ELRS support 14 sensor of telemetry ( this link https://oscarliang.com/expresslrs/ ).
i just want telemetry of  GPS to know where is my gear when video lost signal, etc...

VladimirP1, 2021-04-08:
> Thank you!
> As i know basic telemetry is currently provided (VBAT, downlink/uplink LQ and downlink/uplink RSSI).
> But I saw some picture that ELRS support 14 sensor of telemetry ( this link https://oscarliang.com/expresslrs/ ).
> i just want telemetry of GPS to know where is my gear when video lost signal, etc...

As far as I know, GPS telemetry has been implemented, available in develop branch, see here https://github.com/ExpressLRS/ExpressLRS/pull/204.

comsoft1908, 2021-04-08:
> As far as I know, GPS telemetry has been implemented, available in develop branch, see #204.

yeah, full telemetry is available on develop branch (have not tested it myself as I don't need it). Be aware that with R9 hardware in 868 region I had some difficulties with the develop branch 2-3 weeks ago (see my issue reports) - does not necessarily mean that you'll have the same issues, just to let you know and be careful the first few flights when using develop branch.

s2kaka, 2021-04-22:
Thank for the Infomation! 
Can I flash ELRS through Bootloader now? Or need STLINK and when Flash STlink, can I go back to FRsky stock FW?

Jahnkeanater, 2021-04-22:
> Thank for the Infomation!
> Can I flash ELRS through Bootloader now? Or need STLINK and when Flash STlink, can I go back to FRsky stock FW?

No you need to use a Stlink programmer. This will erase the stock bootloader and you cannot go back to FrSky. FrSky changed it so we cannot create bootloader files for the R9M Lite Pro.

s2kaka, 2021-04-22:
> No you need to use a Stlink programmer. This will erase the stock bootloader and you cannot go back to FrSky. FrSky changed it so we cannot create bootloader files for the R9M Lite Pro.

May have a chance  that r9m lite pro can flash ELRS through bootloader in future? 
A another question is: Are R9mm OTA supported ? i did not see it in the supported rx.

VladimirP1, 2021-04-22:
> May have a chance that r9m lite pro can flash ELRS through bootloader in future?
> A another question is: Are R9mm OTA supported ? i did not see it in the supported rx.

That could be possible if Frsky released a tool to encrypt firmware for their crypto-bootloader on R9M Lite Pro. But I do not think that's gonna happen. Nowadays Frsky is trying to lock you ob their products really hard.

R9mm and R9mm-OTA and R9mini are the same hardware, it is supported.

CapnBry, 2021-04-29:
This has been merged and reported working by users, so I am closing the request to add this target. Excellent work, @VladimirP1!

closed 2021-04-29 by CapnBry

AminZallaghi, 2024-11-25:
Hi all. I tried to flash r9mlp to elrs but in expresslrs configurator i have error : st link not connected