ExpressLRS/ExpressLRS #328: Feature request - configuration via CRSF engine

howels, 2021-01-25:
In original Crossfire and Tracer modules the CRSF interface is used not just for radio commands but also to supply a menu interface to configure the TX module and RXs attached.  
These menus are driven dynamically via the firmware in the TX module and not just via hard-coded prompts in the lua scripts. Because of this an older radio can still use the CRSF engine to configure a newer module with newer features.

Could ELRS use this system also, so that ELRS features could magically appear when the TX was updated and not require new LUA scripts or specific on-radio support for ELRS beyond the existing CRSF support?

expipiplus1, 2021-04-05:
Where/is this part of the crsf specification documented?

howels, 2021-04-06:
> Where/is this part of the crsf specification documented?

The crossfire scripts on opentx are a good place to start: https://github.com/opentx/opentx/tree/2.3/radio/sdcard/taranis-x7/SCRIPTS/TOOLS/CROSSFIRE

expipiplus1, 2021-06-09:
hahah, code as documentation, love to see it

howels, 2021-06-09:
> hahah, code as documentation, love to see it

Well the eLRS code is written and working now so it wasn't that bad. 😉
I did the testing for deviation TX, it's a really neat feature.

closed 2021-08-15 by StonedDawg