ExpressLRS/ExpressLRS #337: R9M 2.4 GHz? Flashing Ghost RX

tew2012, 2021-02-01:
Does the R9M work on 2.4 GHz? It can be compiled (-DRegulatory_Domain_ISM_2400) for the R9M target. I would like to use it with a Ghost rx. I already flashed the rx via st-link utillity, verfied the flash and it was ok, led where on. Since  disconnectiong and reconnecting the St-link v2, the LED is off, the chip is still accessible via the utility, verfied the flash again and it's ok.  Is this normal? When I connect the Ghost rx to a flight controller I get 3.3V on the rx 3.3V pin.

greenbigfrog, 2021-02-01:
r9m is made to work at 900, and only 900.
Only because you can compile with the 2400 regulatory domain doesn't mean it will work.
For 2.4 you'll have to build a (easy) DIY module. (Or wait and possibly buy a ghost tx.)

tew2012, 2021-02-01:
That's what I though. The LoRa chips on Ebay comes with a frequency. I wondered why I could compile it because I expected target depencies preventing it from compiling. I haven't looked into the code yes. Thanks for the reply.

AlessandroAU, 2021-02-03:
IF you wanted to it would be possible to solder in a discrete sx1280 module to enable 2.4ghz support, although it is not really a supported config.

closed 2021-02-03 by AlessandroAU