ExpressLRS/ExpressLRS #340: wikipage Inverter Mod for R9M 2018

Rampetampertje, 2021-02-06:
Can the sentence in the instruction be modified because the bitratemod isn't needed in the X9D!

X9D only does 400 kbit/s and does not allow lowering the bitrate to 115200 bit/s - modification is required 🙅

Only the resistor mod in the R9M 2019 is needed for the X9D.

maybenikhil, 2021-02-10:
The modification that is required is the resistor mod, that's what the page is saying

SilentFlyBy, 2021-04-21:
I'm a bit confused here. If the X9D **only** supports 400k and the R9M 2018 **needs** a hardware mod to support 400k, how does the R9M with the Frsky stock firmware work with the X9D?

MarkusPfaff, 2021-05-03:
On https://github.com/ExpressLRS/ExpressLRS/wiki/Inverter-Mod-for-R9M-2018 under heading "Transmitter Notes" there is a link for the X9D that links to the page by Philipp Seidel where the modifications for other FrSky transmitters are shown. 
This link is missleading and should be deleted.

The X9D does not need any modification. Still the R9M needs the resistor modification to speed to 400kBit/s which the X9D supports.

brandonrc, 2021-05-09:
Updated page. Think it was confusing to have modification is required. Modification is used for both the r9m and the radios. I think that is confusing. 

- `X9D` **only** does 400 kbit/s and __does not allow__ lowering the bitrate to 115200 bit/s - 400k Mod is not required.

https://github.com/ExpressLRS/ExpressLRS/wiki/Inverter-Mod-for-R9M-2018

closed 2021-05-09 by brandonrc