ExpressLRS/ExpressLRS #341: Flight Modes In Ardupilot

lavachemist, 2021-02-07:
With ExpressLRS, I only seem to be able to output three frequencies per aux channel. Is this expected behavior? No matter what I do, I can only seem to output 997, 1499, and 2001 on a given channel. This means that I am limited to three flight modes in Ardupilot, which is a pretty significant downside to using ExpressLRS. Am I missing something?

brandonrc, 2021-02-07:
https://github.com/AlessandroAU/ExpressLRS/wiki/OpenTX-and-Betaflight-Setup#modes

```
By default, ExpressLRS uses one-bit switches for the AUX channels. This means a three-position switch will only send two positions (fully off, or 1000, and fully on, or 2000) to Betaflight on the AUX channels. Set your modes appropriately if you are using one-bit switches, or enable one of the other switch modes in ExpressLRS. There are options that enable 3-pos switches if desired. 
```

lavachemist, 2021-02-07:
I was able to work around this situation at the expense of taking up aux channels. I set a global variable that sets a switch so that the middle position sets channel 8 to 100, and the forward position sets channel 9 to 100. Then I assigned RC8 to an Ardupilot flight mode and RC9 to a different Ardupilot flight mode. This works because Ardupilot will select the most recent mode switch change, overriding all other mode switches (see link below). This is still very inconvenient because it means that you use an entire aux channel for flight modes beyond the three that you can set on the flight modes switch. I hope that this is something that you'll consider changing in the future.

https://ardupilot.org/copter/docs/common-auxiliary-functions.html

norulers, 2021-02-07:
What firmware version supports crsf ？   ArduPlane 4.1？

lavachemist, 2021-02-07:
> What firmware version supports crsf ？ ArduPlane 4.1？

I'm not sure which version of Ardupilot first introduced CRSF support, but I am running Arducopter 4.1.

norulers, 2021-02-07:
@lavachemist  Thank you，SERIAL_PROTOCOL=23  （RCIN）？

lavachemist, 2021-02-07:
> @lavachemist Thank you，SERIAL_PROTOCOL=23 （RCIN）？

Yes, that's correct.

norulers, 2021-02-07:
@lavachemist Thank you again for your help

lavachemist, 2021-02-07:
@norulers no problem. Good luck and happy flying!

CapnBry, 2021-04-22:
Closing this because HYBRID_SWITCHES_8 now supports 2/3/6 position switches.
https://github.com/ExpressLRS/ExpressLRS/wiki/Switch-Modes