ExpressLRS/ExpressLRS #344: feature request: Change FHSSfreqs

iamfaith, 2021-02-09:
Are there any methods to change FHSS frequency by Lua script?

JyeSmith, 2021-02-10:
No.  What is the need?  Just set the correct freq domain for your region when compiling.

iamfaith, 2021-02-10:
For example, while you fly with your friends in 2.4G hz, you may want to avoid interference from each other. I notice there is 1 M hz for the bandwidth so that we can change the channel to have better link quality

JyeSmith, 2021-02-10:
Are you talking about these frequencies?

The hoping order is random and different for every TX, so you already get minimal interference.

https://github.com/AlessandroAU/ExpressLRS/blob/25f66206d0620b122b35329ec9f1e21b633621d1/src/src/FHSS.h#L160

iamfaith, 2021-02-10:
Yes, it would be better if it can be changed via lua script. While flying in the city, there are lots of interference from 2G hz devices.

AlessandroAU, 2021-02-10:
I don't understand what you mean, the protocol randomly hops between channels in the frequency bands. 80 channels on 2.4ghz for example.

iamfaith, 2021-02-10:
> I don't understand what you mean, the protocol randomly hops between channels in the frequency bands. 80 channels on 2.4ghz for example.

[image]
I suggest users can select the channel via Lua script instead of randomly. 
I have tested the project wifibroadcast, if there are fewer devices in the channel, you can have better link quality. If we can switch channels by script, we can select the channel with less interference instead of choosing randomly by the program.

AlessandroAU, 2021-02-10:
I see, you want to have static channel assignment. Unfortunately, this would make it blatantly non-compliant with most regulatorily requirements which have FHSS as a mandate.

iamfaith, 2021-02-10:
> I see, you want to have static channel assignment. Unfortunately, this would make it blatantly non-compliant with most regulatorily requirements which have FHSS as a mandate.

Is there any possibility of adding the advanced setting to have this feature? If not, I will close this issue.

iamfaith, 2021-02-10:
For example, I prefer to use channel 13 in 2.4G Hz. It has less interference while doing video transmitting. And it's legal to use in most of country.

AlessandroAU, 2021-02-10:
That is a little different, that is a wide bandwidth system DSSS occupying some 20mhz of bandwidth. ExpressLRS is a narrow band FHSS approach, it must frequency hop.  If you want to experiment you can define a different range of frequencies e.g here https://github.com/AlessandroAU/ExpressLRS/blob/d231a1dd5a82cd8d205810fd4868940b19031fe8/src/src/FHSS.h#L49

iamfaith, 2021-02-10:
No，i don't mean modify this. I mean if i use 2.4 G Hz, may i choose specific frequency instead of randomly chosen by elrs.

This is an array which i can input the index of this array from lua script to activate frequency.

I don't mean to use custom frequency.

AlessandroAU, 2021-02-10:
I'm not sure you understand how it works, all of the spectrum (~80mhz) is used, every 2nd packet both the RX and TX hop to a new random frequency. You can confine it to a slice of the spectrum if you want, by editing the hop table.

iamfaith, 2021-02-10:
Yes, I was wonder that why not use fixed frequency?

AlessandroAU, 2021-02-10:
Because a single source of interference would knock you out otherwise. FHSS is intentional and an important part of link robustness. 
[image]

iamfaith, 2021-02-10:
Thanks for your clear explanation. Could you please send me the link of this image? I will close this issue later. Sorry for my misunderstanding of intend to change frequency randomly. 
But could we encrypt the data using fixed frequency instead of changing it randomly?

maybenikhil, 2021-02-10:
Encrypting the data doesn't improve the stability of the link (or legality) it just makes the link have extra encode/decode needed.

iamfaith, 2021-02-10:
> Encrypting the data doesn't improve the stability of the link (or legality) it just makes the link have extra encode/decode needed. 

Yep, but it do help in avoiding interception. Anyway, thanks for answering some questions for me. I close the issue now.

closed 2021-02-10 by iamfaith