ExpressLRS/ExpressLRS #345: feature request: Some sort of OLED Display / Joystick for Modules

WillsGledhill, 2021-02-09:
Just had a thought: Would there be a way to implement a OLED display and a joystick into the existing modules? Maybe just soldering to any unused pins on the processor?

ariefadha, 2021-02-09:
Im the way on designing a pcb with oled and 5 way button, but with separate mcu that also handle buzzer.

Im planning with separate mcu because we might need extra features and display so we dont bother with esp32 that handling the rf communication.

JyeSmith, 2021-02-10:
What would an oled be used for that lua can not?  It's also a heap of effort to make a UI.

ariefadha, 2021-02-10:
Deviation does not have, and i think it will be faster to set.

I can make the UI

jfrowies, 2021-03-22:
Everyone has a phone these days, how difficult would it be to interface with an app via wi-fi? I can make the app.

nightflasher, 2021-05-02:
Well, immersionrc ghost got an oled screen - so this seems legit to ask for. I'm in, when buying an esp module, some come with oled screen, so why don't make user of it?
I also have some laying around.

brandonrc, 2021-05-09:
@nightflasher Do you have any experience working with oled stuff? 

I worked on the ghost targets and have both the full and lite. I would love to add support for this.

brandonrc, 2021-05-14:
https://photos.app.goo.gl/GtmHRMXc7JS4nDjE6

brandonrc, 2021-05-15:
Removed the joy stick part.. opening a seperate issue now: 
https://github.com/ExpressLRS/ExpressLRS/issues/603

brandonrc, 2021-05-16:
[image]
Full size 128x64

[image]
Lite size 128x32

These are the ghost tx and ghost tx lite but you can add a OLED 128x64 or a 126x32 to your DIY TX and we can add support for it :)

nightflasher, 2021-05-21:
I've done some work on oled, but mostly:
"foo: $val.foo"
"bar: $val.bar"
And that's all what's needed! IMHO

Maybe another tickbox in the configurator to activate the oled (Btw where can I connect a buzzer/piezo to the ttgo variants?). So you can add an oled if you like to do so and activate it - like the melody or beep.

nightflasher, 2021-05-28:
Today I've started working on this for the ttgo_v2 lora32 with oled screen.
There will be some changes in the OLED defines, and I don't know, if it is okay to prepare the firmware for adding an i2c display for non-oled modules. I would recommend a checkbox and radio button in the configurator to select HAS_OLED and 128x64/128x32. And it should be smarter to program this part to shorten TARGET_TX_GHOST_LITE to HAS_OLED32 and HAS_OLED64 and it needs the ifdef _I2C to differentiate.

I hope everyone understands, been a while since I wrote "deep" technical stuff and so in english.

brandonrc, 2021-05-28:
@nightflasher if you can use my branch pr-oled-support then make a merge request to that branch I would love to check the changes out!!!

I agree with your changes.. we do something smart where if 
```
#ifdef GPIO_PIN_OLED_I2C
Do xyz for i2c stuff
#else
Do oled stuff for SPI
#endif
```

So just based on which pins you are using we can know wether it is i2c of spi. :)

brandonrc, 2021-05-28:
@nightflasher I just ordered a ttgo_v2.1 to help out something with debugging issues. I could also test your code when you make the pull-request to the `pr-oled-support` branch.

nightflasher, 2021-05-29:
i'm already working with this branch, trying first time platformio, but doesn't seem to be to hard.

closed 2021-07-17 by brandonrc