ExpressLRS/ExpressLRS #352: Switch survey

JyeSmith, 2021-02-17:
We regularly have switch discussions on discord and it is probably worth while adding a couple more options.

The current default switch configuration is 4 2-position switches, plus we have the option to select HYBRID_SWITCHES_8 for 8 3-position switches.  We could add a couple more options similar to hybrid switches e.g. an arm switch that is always sent plus the others that are cycled through round robin. 

Technically we have 7 bits to play with and below are a couple of suggestions.

- 2-way arm + 4 16-way
- 2-way arm + 8 8-way
- 2-way arm + 16 4-way

maybenikhil, 2021-02-17:
Because of the ardupilot 6 way concerns, I think 2 + 8 8-way, and replacing hybrid-8 with 2-way+16-4

JBKingdon, 2021-02-17:
Re Nikhil's comment, it would be nice to keep the existing hybrid-8 if possible, I find the 3 pos arm switch useful.
2-way arm + 8 8-way looks like a very versatile addition, we'd need to confirm that it meets the inav need, i.e. that their config is flexible enough to cope with the positions.
I'm not sure that crsf protocol supports >16 channels does it? So the 16 4-way option may be a bit wasteful.
The 4 16-way might be useful if that's enough resolution to meet some of the extra control requirements - flap positions maybe? Perhaps it's enough for a P/T camera gimbal?

0crap, 2021-02-17:
The current situation with HYBRID_SWITCHES_8 suits my needs just fine with Betaflight and iNav.
Please make sure this stays possible in the user_defines.
So no suggestions here for a change. :-)

howels, 2021-02-17:
I would normally use 2x 2-way and 6x 3-way switches. Overkill for most of my models but there are some where additional camera control (runcam start/stop record and menus) and other functions use a bunch of switches.

ZzyzxTek, 2021-02-18:
2 pos Arm + 4 X 3 pos + 1 X 6 pos would work for what I currently do.

brandonrc, 2021-02-18:
2-way arm + 8 8-way sounds perfect. 

I really like this. If someone wants to use a 3 way arm switch then they can modify the open tx model to have that physical 3 position switch work with the 2pos limitation.

dwhacks, 2021-02-18:
So far the hybrid switches have worked great for me, but the 2-way arm + 8 8-way sounds like it could be very useful also.

gnappo72, 2021-02-18:
Ciao Jye HYBRID_SWITCHES_8 suits me well, however other options r always welcome as it's always difficult to please everyone...

hydra, 2021-02-18:
Just thinking a bit out of the box, but wouldn't it be nice to assign priorities to switches in the OpenTX firmware, priorities then communicated somehow to the ELRS TX and it just does what is appropriate.  i.e.  send high priority switches often, medium priority switches less frequently and low priority switches even less frequently.    Granted, OpenTX isn't the only TX system that ELRS works with, but food for thought.

phobos-, 2021-02-19:
What about the original option of 8x 1 bit switches sent always? Is that going away? For me that is the preferred option, any 3 or even 4 pos switch can be converted to 2x 1 bit switches.

P-I-Engineer, 2021-02-19:
my vote

2-way arm + 8 8-way

i like to do replaces on opentx for using multiple switches to move a single channel around.

Dropsleutel, 2021-02-20:
What about an alternating mode, alternatively fill the packets with different info ? F.i. double the amount of channels and switches, but at costs of half of the update rate ?  1-4  /   5-8  ? Not intend for the pure racer, but for aircaft with need for some more channels ?  (possibly 5-8, could even be alternated with 9-12, that will half the refresh rate compared to the main control channels.

expipiplus1, 2021-03-20:
Is it beyond the realms of possibility to expose this as configuration to the user? For instance have a little DSL/GUI for them to use to describe how to pack and transmit their inputs.

enrico1036, 2021-03-26:
Why not 8 16way? Arm would be one of the 8 round-robin channels. I don’t think arm switch need to be so fast. Even at 50hz it would be sent every about 160ms. Not a big problem if you compare to human reaction time. 
Users should be able to choose between options.

JyeSmith, 2021-03-26:
@enrico1036 ExLRS started with a separate switch packet and the disarm delay is real and concerning.  So the 1b option was added to send switches all of the time.  Even now the sequential switch option still exists and it round robins the arm switch, but this option isn't visible to users.

I think we all see it as a serious safety issue not to send the arm switch with every packet.  What happens with low rssi/lq when it is now possible to miss switch packets.  Particularly users using low RF modes.  Disarming can now take a significant time.

enrico1036, 2021-03-27:
Yes, you’re right! I was not considering lost packets. 
What would be the down sides of adding an entire byte to the packet? Maybe someone could choose to have more channels with high resolution at the cost of some more latency. For example to move a cam on a gimbal

Dropsleutel, 2021-03-27:
yup exactly what I meant, for some application maybe more channels and switches at cost of latency. 
4 channel mode fastest. 
8 channel mode double the delay.
12ch mode,  channel 5-12 tripple the delay.  
Is there still some room to gain by encoding into bcd f.i. ? instead of separate bits ?
Sending only arms switch always may be a good idea for safety ....... 

However all those switches setup etc, still very old fashioned.   Would be fantastic if in the Tx you can load a lcommand ibrary, f.i. Betaflight. You go to a switch in the model setup,  assign f.i. ARM, command or whatever instruction.  In Betaflight nothing to setup it should just get the instruction send from Rx and apply.

CapnBry, 2021-04-11:
The reason you can't "just add more bytes" to the packet is because the amount of time it takes to transfer the packet over RF defines the link packet rate. Sending the current RF packet for 250Hz takes 3394us out of the 4000us of air time available. 9 bytes would be 3820us, but that doesn't include all the overhead and jitter so you'd start to run into trying to start a transmit while the previous packet was still going out.

I love HYBRID_SWITCHES_8, I think it is a great compromise over 4x 1-bit switches while still fitting in the size requirement. I can see the desire for:
1x 2-position ARM (always sent), 4x 3-position, 2x 8-bit channels for pan/tilt

I don't think that will fit any way we slice it though, considering there are only 7 bits **total** to work with. Maybe of the YAW channel was split so 6 bits were used for YAW and 4 bits were used to extend the round robin channel (to 6 bit). I think most folks who want wider AUX channels are doing it for fixed wing so maybe having a reduced resolution yaw is acceptable, however only 64 possible yaw values does seem low. 64 possible pan/tilt values is pretty ok I think. It would be jerky but you're not trying to splice a fiber optic cable, just aim a camera in the general direction of what you want to look at.

I've also considered splitting an 8-bit value across 4x 2-bit cycles, but that gets really dicey because you can't report the value until you know you've gotten all 4 pieces, and the value isn't changing. Maybe accumulate the value and if the same value is received across cycles, report that to the FC? I feel really uncomfortable about that.

Dropsleutel, 2021-04-11:
yes encoding BCD makes much more channel, so 4 bits, 8 switch positions, if can add one bit so 5, we can go to 16 switch positions ?
Another methot can be to "circulate/multiplex" switches channels?  But that will increase latency but should not be issue ? maybe multiplexing it can be synced locked to the frequency hopping ? Maybe the ARM switch, can be kept out and send allways ?

pkendall64, 2021-04-11:
With the recent changes to `develop` the OG switch mode is now 4 10-bit channels and 8 1-bit channels.

I think there is room for another switch mode. One that has 4 10-bit channels and 3 3-pos and 3 2-pos channels. That can be encoded into the 8bits that the OG mode used for the 8 1-bit switches.

Dropsleutel, 2021-04-11:
Than all problems solved isnt't it ?  8x1 bit channels,  you can encode 500 on/off  in BCD ?

CapnBry, 2021-04-11:
You keep saying BCD but BCD takes more space than binary (only 10 of the 16 possible values of bits are used). If you wanted to switch the 8 bits to send one 8-bit channel, then you only end up with 4x 10-bit (sticks) and 1x 8-bit. I don't think anyone would want that.

The current implementation uses 2 bits for a 3-position switch or 1 bit for a 2-position switch. I'm not sure you'd be able to pack 3x 3-pos + 3x 2-pos, but 3x 3-pos and 2x 2-pos is possible I think? HYBRID_SWITCHES_5?

Another problem is that the layout is fixed at compile-time, which royally stinks for someone who wants to be able to control a quad and a fixed wing, since you need to decide and bake in your channel layouts for **every** RX you own. It is currently impossible I think to fit both channel implementations in IRAM, so that's even more of a problem the more switch layouts are supported.

pkendall64, 2021-04-11:
You can get 3x3-pos and 3x2-pos if it's not binary encoded. i.e. 2 bits for for 3 pos wastes a 1/2 bit. If you do it via multiplication (slightly slower admittedly) you can fit them because 3*3*3*2*2*2 = 214, which fits into 8 bits.

CapnBry, 2021-04-11:
Oh wow, I figured it would be possible somehow, but I do not understand how that math works! I feel like HYBRID_SWITCHES_8 covers that same style though, even though they're updated more slowly, they would practically feel the same for the user.

padcom, 2021-11-28:
@JBKingdon I am assuming you're using the 3-pos switches for example for arming with autolaunch or for switching 3 basic modes (like manual, acro, 3d cruise). In that case you can use the 2-pos switch for RTL which (in long-range) is the most important information by any standards. You can land and not be able to disarm for a second or two but if your model flies away and you don't know your way home the RTL mode is there to save your model.

pkendall64, 2022-02-04:
Can't we close this now with Wide switch mode et.al.

closed 2022-02-10 by JyeSmith