ExpressLRS/ExpressLRS #381: Support for SIYI FM30 TX

CapnBry, 2021-03-13:
The SIYI FM30 is a 2.4GHz transmitter module for JR bays based on the SX1280 chip with +22.5dBm output power (via SiGe 2431L) and integrated bluetooth. The MCU is an STM32F103C8 with a 16MHz crystal (HSE_VALUE=16000000U). Flashing is done via STLINK through a conveniently labeled SWD pin header. I'm working on support but I'm creating this ticket to provide the pin mapping and layout for interested parties.

| Pin | Function | Remarks
|---|---|---|
|PA0| 16 pos switch | Bit 0001
|PA1| 16 pos switch | Bit 0010
|PA2| USART2_TX | Bluetooth TX
|PA3| USART2_RX | Bluetooth RX
|PA4|16 pos switch| Bit 0100
|PA5| 16 pos switch | Bit 1000
|PA6| PPM Input? | Top pin (1) of JR module, labeled SBUS RX3 PPM, through diode
|PA7| LED Left Green | Active low
|PA8| Bluetooth VCC Enable | High to enable Bluetooth chip's power
|PA9| USART1_TX | RC Signal TX, connected through XOR chip / diodes to JR pin 5
|PA10| USART2_TX | RC Signal RX, connected through XOR chip / diodes to JR pin 5
|PA11| USB |
|PA12| USB |
|PA13| SWDIO |
|PA14| SWCLK
|PA15| LED Left Red | Active low
|PB0| Button | Active low
|PB1| LED Right Green | Active low
|PB2| LED Right Red | Active low
|PB3| nRST |
|PB4| 2431L Antenna Select | Low for pre-installed antenna (Left IPEX)
|PB5| RX3 Inverter? | 
|PB6| ???? | Goes to *somewhere* but can't find other end
|PB7| TX1RX1 Inverter | BUFFER_OE
|PB8| SX1280 DIO1 |
|PB9| 2431L CTX | High for transmit, Low for receive
|PB10| USART3_TX | Not connected (missing resistor to connect to JR pin 2)
|PB11| USART3_RX | Connected through inverter to JR Pin 1 (shared with PPM)
|PB12| SX1280 SPI chip select |
|PB13| SX1280 SPI SCK |
|PB14| SX1280 SPI MISO |
|PB15| SX1280 SPI MOSI |

The RF module has pins brought out to castellations. One of the bottom left pins may not be ground, but it is connected directly to ground.
| Left | Right |
|---|----|
| | SX1280 RST
| | SX1280 DIO1
| | SX1280 SPI chip select
| | SX1280 SPI MOSI
| | SX1280 SPI MISO
| | 3.3V
| 2431L CTX | 
| SX1280 Busy |
| 2431L Antenna Select |
| SX1280 SPI SCK |
| GND |
| GND |

### Challenges
* No BUSY pin connected to MCU. ELRS TX works if you bypass the BUSY pin checks but I'm not sure how this will affect stability.
* No external EEPROM. Configuration will need to be stored in STM32 flash, possibly using [AN2594 EEPROM Emulation](https://www.st.com/resource/en/application_note/cd00165693-eeprom-emulation-in-stm32f10x-microcontrollers-stmicroelectronics.pdf)
* The RF amplifier is fairly limited, in that you can only transmit at full power or not at all. Input power is limited to +6dBm.
* Bluetooth would be sweet to add MSP support, but the chip label has been lasered off. Possibly Zhuhai Jieli Technology Co., Ltd. AC6905A or similar? 24pin QSOP.

Try to ignore my flying wires here, I forgot to take photos before I started.
[image]

[image]

Dropsleutel, 2021-04-09:
Is there any hope that we can have the SiYi Rx ported to operate as a 2.4ghz Tx module ?  (A similar thing has been done with the R9 Rx at the time)
SiYi has the same cpu and RF chain, inc 250mW Pa. As it is super tiny it would be really great for embedding inside my tx. 

https://github.com/ExpressLRS/ExpressLRS/issues/308

CapnBry, 2021-04-10:
I'm closing this because this is merged now. SIYI RX as a TX will come as a new PR if it is done.

closed 2021-04-10 by CapnBry