ExpressLRS/ExpressLRS #389: Lua script does not show data and can not save settings.

Noctaro, 2021-03-16:
Hi all,
i have troubles using the LUA script for monitoring and adjusting settings on my Frksky X-Lite. After i have tested it with OTX 2.2.3, i decided to update my X-Lite to OTX 2.3.10 from the ELRS repo. I am using the TTGO V2 transmitter. The LUA script will always work the first time after changing the baudrate in the hardware tab in OTX. It does not matter what baudrate option i choose. If i close the LUA script, i again have to change the baudrate before i get data from the script again. If i change settings within the running script, they will not be stored. It seems to me there is a problem on how OTX handles the script. Is there anything i could have done wrong?

GhVost, 2021-03-20:
I can only confirm the issue. I think the lua script instead of doing the right thing crashes the program of the module.

GhVost, 2021-03-20:
I have question: why did you decided to use the "-DRegulatory_Domain_EU_868" in your 2.4 GHz module?

Noctaro, 2021-03-20:
> 
> 
> I have a question: why did you decided to use the "-DRegulatory_Domain_EU_868" in your 2.4 GHz module?

I do have a 868mhz module, not 2.4ghz :)

Noctaro, 2021-03-26:
I have tested my TTGO V2 using the dev branch today, LUA script is working now! :)
Should i close this issue?

GhVost, 2021-03-27:
Great! Can you share the link?

Noctaro, 2021-03-27:
> 
> 
> Great! Can you share the link?

https://github.com/ExpressLRS/ExpressLRS/tree/develop

GhVost, 2021-03-30:
Hey,
thanks a lot! I use the dev branch since then without any issues. Do you know by any chance when the battery voltage and other important things will be added to the telemetry parameters?
Please close the issue, but add the solution in your last comment — the elrs is just picking up and it will definitely help hundreds.
By the way, I had this issue with the 2G4 DIY module.
 
Cheers

> 
> 
>I have tested my TTGO V2 using the dev branch today, LUA script is working now! :)
>Should i close this issue?

closed 2021-04-03 by AlessandroAU