ExpressLRS/ExpressLRS #390: Support for Telemetry view App

kusnendar, 2021-03-17:
I hope ELRS can use Telemetey Viewer App Android/IOS. Telemetry dta transfered from tx to smartphone via smarport + bluetooth module.
Like this: https://youtu.be/lmSMt6JLWL8

CapnBry, 2021-03-24:
I was actually playing with this yesterday using the SIYI FM30 which has integrated bluetooth. I modified my RX code to just copy the CRSF telemetry over to the bluetooth (and verified I saw data on my phone) but the Telemetey Viewer app didn't like it. I think it had to do with it trying to auto-detect the protocol and there not being enough valid telemetry in the short time it probes to meet the minimum auto-detect threshold (2 complete sets of telemetry? Link Statistics don't count).

Once the SIYI TX and RX pull requests go through, I'll look at working out with the devs what ELRS will put out as standard and then maybe work to integrate it. The app supports MAVlink v1 and v2, LTM, FrSky S.Port, and CRSF. CRSF would be the easiest since we've already got the data in that format but MAVlink would probably be more compatible.

kimman1, 2021-03-30:
> I was actually playing with this yesterday using the SIYI FM30 which has integrated bluetooth. I modified my RX code to just copy the CRSF telemetry over to the bluetooth (and verified I saw data on [...]

Can you show me your modify code for copy CRSF tlm ?

s2kaka, 2021-04-08:
hope it will support this app with build in Bluetooth like x-lite pro, t16, t18.

avsaase, 2021-05-04:
I can confirm this is already functional on a TX16S with a HC05 module connected to the UART.

TarasTadva, 2021-05-20:
Tell me! I have a Jumper t12 transmitter with a FrSky r9m module. There is no built-in UART in the Jumper. How do I use UART in a module? There is a S-port output, and bluetooth telemetry works through it in the FrSky, but after the firmware the Express Lrs does not work (((((

nightflasher, 2021-05-29:
Nice start, BUT: make also use of the Bluetooth part of the esp32 modules (mostly ttgo_v1 and ttgo_v2).
Maybe there is someone who can program an smartphone app for changing settings in the module (for those who can't run Lua, like modified or homebrew transmitters) - ttgo T-Watch-2020.

hyperion11, 2021-12-02:
hi! it possible to sent telemetry stream via native s.port pins as stock R9 fw do?
[image]
HC-06 with inverter
[image]

CapnBry, 2021-12-02:
I believe it is. I think you can also use EdgeTX Telemetry Mirror without an inverter too, on handsets that have a serial port (like the TX16S).

In 2.0, there's also an option to turn on `BT Telem` in the Lua options for FM30 TX modules. In 2.1 TX modules with backpacks will be able to use wifi mode and connect using a TCP/IP socket connection for a mirror.

hyperion11, 2021-12-03:
I check R9M with this connections. No telemetry. I will try solder BT to pads on QX7 Motherboard and flash to QX7S FW. And check again... it would be nice to use S.PORT pins on R9M module to send telemetry stream...

brandonrc, 2021-12-19:
@CapnBry so as of `2.1` we should have telemetry view wifi to an app?

CapnBry, 2021-12-20:
> @CapnBry so as of `2.1` we should have telemetry view wifi to an app?

That's the plan. To make it so when the backpack is in wifi mode it has a port open to stream raw telemetry to (and accept config commands too).

karl03, 2022-01-15:
>
> That's the plan. To make it so when the backpack is in wifi mode it has a port open to stream raw telemetry to (and accept config commands too).

Will this be Mavlink compatible?

bike797, 2022-02-14:
It would be nice to be able to connect with the android telemetry app using the BLE of the esp32 backpack.
 - https://openlabpro.com/guide/ble-uart-on-esp32-controller/
 - https://github.com/CrazyDude1994/android-taranis-smartport-telemetry

pkendall64, 2022-02-14:
> It would be nice to be able to connect with the android telemetry app using the BLE of the esp32 backpack.
> 
> * https://openlabpro.com/guide/ble-uart-on-esp32-controller/
> * https://github.com/CrazyDude1994/android-taranis-smartport-telemetry

The backpack is an 8285 which does not have BLE

sweebee, 2022-05-08:
@CapnBry any news on this? 😇

CapnBry, 2022-07-06:
That's a great bunch of description, but you don't need to do quite all that. All you need to do is one line to mirror everything to the backpack UART (and change the baud):
`crsf.PortSecondary = TxBackpack;`

GengKanghua, 2022-07-07:
> Is this all really necessary? I ask because I use the same application, without any code or backpack changes, just using the USB port of my Radiomaster TX16S with EdgeTX 2.7.

You are right, it is not necessary for TX16S, but I don't have TX16S R9M or FM30. This is my solution for my existing equipment（TX12）.

ASDosjani, 2022-08-22:
Is anyone working on this? It would be good if radios with built in elrs work. The USB of the radio could be used to transmit telemetry data.

dz0ny, 2022-08-22:
Yes this is attempt to use BLE on esp32 based tx https://github.com/dz0ny/ExpressLRS/tree/ble-telemetry

dz0ny, 2022-08-22:
> Is anyone working on this? It would be good if radios with built in elrs work. The USB of the radio could be used to transmit telemetry data.

You can already do this in edgetx, VCP mode under Hardware needs to be set as Telemetry Mirror.

Lance0724, 2022-10-19:
> 

It's cannot use with android-taranis-smartport-telemetry? The problem is in android app?
Can I use in U360GTS?
I want use AAT with elrs.

Lance0724, 2022-10-19:
> > > This is attempt to use BLE on esp32 based tx https://github.com/dz0ny/ExpressLRS/tree/add/ble-telemetry
> > 
> > 
> > If merge that code, can I use telemetry view in my phone or some AAT ?
> > Has this been tested?
> 
> It's missing initial payload that _Telemetry view App_ uses to determine supported protocols, if you add that then it can be used yes.

For now It's cannot be used with android-taranis-smartport-telemetry? The problem is android app?
Can I use in U360GTS?
I want use AAT with elrs.

dz0ny, 2022-10-19:
> For now It's cannot be used with android-taranis-smartport-telemetry? The problem is android app? Can I use in U360GTS? I want use AAT with elrs.

No my patch needs a bit more work :) Without additional work it cannot be used.

Lance0724, 2022-10-19:
> No my patch needs a bit more work :) Without additional work it cannot be used.

Out of the blue, do you have plans to complete it?
I'm really looking forward to this feature
I don't want to add a bluetooth module to my transmitter

dz0ny, 2022-10-19:
> Out of the blue, do you have plans to complete it? I'm really looking forward to this feature I don't want to add a bluetooth module to my transmitter

When time permits :) Other developers also suggested this should be moved to backpack to prevent for it to interfere with critical code used in TX.

Lance0724, 2022-10-19:
> 

It would also be nice to move this feature to the backpack

pranas, 2022-11-07:
I'm also interested in a "Telemetry Receiver" backpack. I would like a firmware for RXs that would allow them to bind, receive telemetry and spit it out via serial using CRSF protocol. This would enable building a lot of interesting things like an antenna tracker backpack. I guess I will have to get myself familiar with the codebase 😃 Is there anyone who could give me a jumpstart on how to approach this?

derFrickler, 2022-12-23:
Thanks for this thread! I just connected a HC-05 with the inverter transistor mod to the DIY 2.4 ELRS module in my QX7 and i directly get telemetry data in the App without needing to change anything in OTX.  As i had already set the CRSF protocol to 115200 in OTX.

RomanLut, 2022-12-25:
Finsihed implementation and submitted pull request  https://github.com/ExpressLRS/ExpressLRS/pull/2018

RomanLut, 2022-12-25:
> Can anyone tell me if this will work with the DIY module in my Jumper T-Lite aswell? What baudrate does the T-Lite use for the external CRSF Module?
Well as you said it uses speed  which you set in opentx. And you have to set it to 115200 because HC-05 can not handle default 400000 baud.

RomanLut, 2022-12-25:
> It's missing initial payload that _Telemetry view App_ uses to determine supported protocols, if you add that then it can be used yes.

Telemetry viewer app does not need any initial payload. It will start decoding once 3 successull CRSF packets are received.
Actually the problem was that TELEMETRY_CRSF_UUID = 0x2A4D is some unlucky UID number - nothing is transmitted. If you change it, everything works fine.

derFrickler, 2022-12-26:
> Well as you said it uses speed which you set in opentx. And you have to set it to 115200 because HC-05 can not handle default 400000 baud.

There is no option to set it in OTX on the T-Lite, thats why i was asking if somebody knows the used speed.

Great that it will work even without the HC-05 with your patch, can't wait to see it included.

RomanLut, 2022-12-26:
> There is no option to set it in OTX on the T-Lite, thats why i was asking if somebody knows the used speed.
I thought latest opentx has is, but I did non try it myself. Flashed EdgeTx, there is an option to set crsf baud rate.

derFrickler, 2022-12-27:
YES, baud-rate option appeared after updating to the latest OTX version. Will try edgeTX on my backup T-Lite

derFrickler, 2022-12-29:
Short feedback, hc-05 with the transistor inverter and module and otx set to 115200 baud works just fine.
Now I've got a Jumper T-Lite with 5in1 and BT-telemetry, all build-in 👍

endymuhardin, 2023-08-13:
Hopefully #2305 will finally put this issue to rest :D

dang-itt, 2023-08-21:
Is it possible to connect the HC05 module with inverter to the JR bay S-Port? That way it would constantly be sending S-port telemetry which is what I assume the R9m module does anyway. (I assume the R9m module S-port is just passthrough of the radio s-port)

AlanAlger001, 2023-09-27:
If used in gemini mode, the hardware will keep restarting.

Aprstyo11, 2024-11-16:
> Short feedback, hc-05 with the transistor inverter and module and otx set to 115200 baud works just fine. Now I've got a Jumper T-Lite with 5in1 and BT-telemetry, all build-in 👍

Can you show me how to do that?