ExpressLRS/ExpressLRS #397: reception issues with latest develop branch (EU_868, R9 hardware)

comsoft1908, 2021-03-19:
I have come across very strange reception issues with the latest develop branch (actually already for quite a while, but thought its due to my hardware or setup).

I have cross checked everything (hardware, antenna, settings) and tested with all combinations of hardware I have on multiple quads with analog and digital VTXs, hardware is: R9M (on QX7 ACCST with mods and expresslrs OpenTX version) and R9M Lite, R9M Lite Pro (on X-Lite pro and Tango2) against multiple R9mini RXs and multiple R9slim plus RXs.

I also have cross-checked against a setup with Version 1.0.3 RC2 (for which I backported the R9MLP branch so that I can also test with the R9M Lite Pro). The setup with Version 1.0.3 RC2 is reliable on all hardware mentioned above in every possible combination and under the same test conditions as described in the next paragraph.

My setup is:

* EU_868
* bindingphrase is set
* following switches are enabled, everything else is off/disabled:
  * HYBRID_SWITCHES_8
  * FEATURE_OPENTX_SYNC
* I did all my test with 
  * 10mW, 25mW and 50mW
  * 200Hz mode
  * 1/64 telemetry ratio
* test were:
  * walking downstairs 2 levels in my house and behind some concrete walls and covering antenna with hand
  * flying in my garden, ca. 70m max distances with some concrete walls between me and quad
  * flying outdoors, max distance 200m, clear LOS between quad and TX

Now here are the issues I have with the latest develop branch:
* LQ is jumping very much (in "walking test" it jumps between 100 and 40 and connection loss) - min LQ of 1.0.3 RC2 in exact same conditions is 88, no connection loss. The RSSI values are very similar between develop and RC2 under similar conditions, though.
* when flying outside it sometimes works ok (not great, LQ jumps a lot, and I did not go below 50mW outside), but sometimes I have FS in very short distances (like 100m or less)
* I can't get a reconnect with the develop branch when I switch off the TX and switch it on again, however, with RC2 that works flawlessly under the same conditions

I have tried to find a reason for this behavior looking though the code but there are so many changes that its very hard. What I can say is that it started to be unreliable even before the telemetry PR was accepted.

I am using version 1.0.3 RC2 and my backport for the R9MLP (that I will also publish on my github repo asap) for the time being, but it would be really good to find the root cause of these issues.

Please let me know what steps / tests I should perform to help find the root cause.

digaus, 2021-03-23:
Since I also have the R9M Lite Pro and R9M Lite I am interested in the backport. 

Did not fly with the latest version yet but if it seems to have issues I would like to try the backported RC2 aswell.

zend1ng, 2021-03-24:
Im also having an issue with R9M tx 2019 module and my taranis qx7s with baudrate mode. I flashed my Tx and all my Rx mostly r9 minis and mms and went out today do to a testflight. It was working great and the link Quality was staying between 95-99 but then all of a sudden i get a failsafe and fall out of the sky this happened to 2 different quads with 2 different Rxs twice. i wasnt rly far out so that cant be an issue i tried 10-100mw but no change. Discord told me to try and reflash the tx with esp enabled i havent got the time to test it agian now. Also before the flash i had 1-2:199 in the lua script on my taranis now after the Flash its always staying at 0:200-201 im using 400k baud but also got the issue with 115k ist this issue known or are there any fixes Help would be greatly appreciated since after i flashed all my r9 hardware i dont have anything else right now. 
I feel like it must be some kind of bug or so since its working fine with great link just until it doesnt and randomly failsafes.
Cheers keep up the good work

comsoft1908, 2021-03-29:
> Since I also have the R9M Lite Pro and R9M Lite I am interested in the backport.
Sorry, I've been away for a few days. Will create the branch on my repo tonight, will post a link here.

comsoft1908, 2021-03-29:
> Discord told me to try and reflash the tx with esp enabled i havent got the time to test it agian now.
I assume this will fix the issue for you. 
Easiest way to flash the RXs and TX is the ExpressLRS Configurator. I would recommend to use version 1.0.3 RC2 for both the RXs and the TX, this version is working great for me so far.

digaus, 2021-03-30:
> > Discord told me to try and reflash the tx with esp enabled i havent got the time to test it agian now.
> 
> I assume this will fix the issue for you.
> Easiest way to flash the RXs and TX is the ExpressLRS Configurator. I would recommend to use version 0.1.3 RC2 for both the RXs and the TX, this version is working great for me so far.

Thhis option is not available for the R9M Lite Pro :/

nangel0, 2021-03-30:
Did you have done the resistir mode? What hapens with your uart number in the lua script? I have nailed down my problems to uart instability caused by heat. Diving deep in discord I found many references to this,  even a user that had logged the temperatures until uart failure.

Now I have stable functioning with a 330 ohm resistor, previously was using a resistor too high (3.7k ohm)

bernii, 2021-04-13:
@comsoft1908 any luck with creating the backport branch for R9M Lite Pro? Seems I'm having similar issue in the EU region - would love to verify your fix :)

bernii, 2021-04-16:
nvm
seems to be working OK with the latest updated to `develop` head

nangel0, 2021-04-17:
LQ fluctuations seem to be corrected with 5f61ecf [https://github.com/ExpressLRS/ExpressLRS/commit/5f61ecff771664482961c4d50198f02ccedd0539](url)

can we close this?

comsoft1908, 2021-04-19:
@nangel0 Thanks for the notice. I'll give it a try on the weekend and then can hopefully close this.

comsoft1908, 2021-04-21:
Have tested with develop branch from today: issue seems fixed.

Great work! Thank you very much!

closed 2021-04-21 by comsoft1908