ExpressLRS/ExpressLRS #421: PPM or S.BUS input for TX module.

railbox, 2021-03-31:
Hi,

Is it possible to add PPM or S.BUS input support for TX module? Im thinking to use it as a 2.4G -> 900M converter from other TX.

theproff777, 2021-04-13:
I second this option as I was considering using a Crossfire module with my tx, which I know works, but would much more prefer to go ExpressLRS instead.

devanesanandrews, 2021-04-13:
> Hi,
> 
> Is it possible to add PPM or S.BUS input support for TX module? Im thinking to use it as a 2.4G -> 900M converter from other TX.

it will be great to have PPM which means you can use ELRS with any cheap transmitter

marianomd, 2021-05-29:
From AlessandroAU on RCGRoups: 
[Officially not a supported configuration, but the code is there.](https://www.rcgroups.com/forums/showthread.php?3437865-ExpressLRS-DIY-LoRa-based-race-optimized-RC-link-system/page4)

I've been looking at the source code and haven't find any PPM code. Maybe it's in some branch?

This would be great to use with FlySky FS-i6x with the JR module bay mod for example.

marianomd, 2021-06-17:
I'm [porting OpenTX to FlySky i6x](https://github.com/marianomd/opentx) to be able to support ExpressLRS modules on that radio.

anadezhd, 2021-07-28:
> I'm [porting OpenTX to FlySky i6x](https://github.com/marianomd/opentx) to be able to support ExpressLRS modules on that radio.
In general this will resolve my issue.

railbox, 2021-08-11:
Can someone give me some hints(Code lines) how to introduce S.Bus input support and I will try to prepare the code changes as Im still thinking this is the best solution for me as I want to be R9M module alone on tripod with some S.Bus receiver.

closed 2022-01-21 by schugabe

dhilung, 2024-04-11:
Looks like this is happening #2528