ExpressLRS/ExpressLRS #426: Add Regulatory_Domain_India_866 ?

saidinesh5, 2021-04-01:
India currently delicensed the 865-867 MHz band with a maximum of 1W Transmitter power, 4Watts Effective Radiated Power and 200Khz carrier bandwidth as per https://dot.gov.in/sites/default/files/Delicensing%20in%20865-867%20MHz%20band%20%5BGSR%20564%20%28E%29%5D_0.pdf , which is a subset of frequencies of `Regulatory_Domain_EU_868` . Would you accept a pull request that introduces a `Regulatory_Domain_India_866` config option, that provides these frequencies?:

```
const uint32_t FHSSfreqs[] = {
  865000000
  865200000
  865400000
  865600000
  865800000
  866000000
  866200000
  866400000
  866600000
  866800000
};
```

AlessandroAU, 2021-04-01:
Sure, submit PR and I will approve it

saidinesh5, 2021-04-01:
Btw. what should the spacing between the frequencies be? I used 200Khz because that's the legal maximum carrier bandwidth but 
I noticed that the `Regulatory_Domain_EU_868` frequencies are spaced 525000 Hz apart.

AlessandroAU, 2021-04-01:
The bandwidth of the lora signal is 500khz so you'll have to check the legal requirements for that in India. Usually it will fall under DSSS (Direct-sequence spread spectrum)

saidinesh5, 2021-04-01:
I will check.
So far the information specifically on DSSS is hard to find in the documents here but they did release this SX1262 chip specifically for india: https://www.semtech.com/products/wireless-rf/lora-transceivers/sx1262dvk1das . 
So starting at 865275000 (like EU did at 863275000) should leave us with these 4 bands.:

```
const uint32_t FHSSfreqs[] = {
865275000
865800000
866325000
866850000
}
```

closed 2021-07-18 by AlessandroAU