ExpressLRS/ExpressLRS #429: Frsky R9M Heat Issues Root Cause

Dropsleutel, 2021-04-02:
The R9M needs to be kept cool, so for power level above 200-250mW a fan is recommended.
The symptoms of too hot R9M, are data loss, dropped packets it is visible in the OpenTx Lua and should be 200packets no loss. A hot R9M is potentially dangerous and crashes have been reported.

Although cooling works, it is not clear what is the Root Cause of the dataloss at higher temperatures.

According some discussion with JamesK, the loss of messages from R9M it seems not the RF link itself which failes undr temperature, but seem to be caused by serial IO errors between R9M and OpenTx.

If somebody has schematic of the R9M serial and buffering, please post here rather than connect a fan like to find the Root cause.

token9, 2021-09-29:
This bothered me as well, and I've gone to town on this, so sharing what I have learnt. No Schematics but looking at the boards its quite clear to me what is going on. 

I just took the dive into ExpressLRS over the past few days, man its been an exciting journey. 
I've moved my r9M Tx and r9slim+ Receiver to ExpressLRS. 
And yes I noticed it ran warmer even at less than 100mW Tx power. What about the new firmware causes the heating? because its the same hardware!

Here is the result of my investigation: 

The [Semtech SX1276](https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R0000001Rbr/6EfVZUorrpoKFfvaF_Fkpgp5kzjiNyiAbqcpqh9qSjE) chip in the r9m Tx cant pump out a lot of power on its own. It has two RF outputs - 

RF_LO_output - this is not used. (efficient because optimised for High UHF bands but can only output 25mW ) 
RF_HI_output - this is not used. (efficient because optimised for Low UHF bands but can only output 25mW ) 
*RF_PowerAmplifier_output - (not very efficient because wideband but can only output 100mW ) 

This 100mW is the reason telemetry data back from the receivers (like the r9slim+) is limited to 100mW.

But on a Tx, we would like more than 100mW. The solution has been to introduce an amplifier - On the r9m Tx it is the [SKY65111 RF amplifier](https://www.skyworksinc.com/-/media/SkyWorks/Documents/Products/101-200/200428E.pdf). All control signals must pass through this amplifier. 

Looking at the specification, it is only about 40 to 50% efficient. Meaning if its pumping out 1W of RF power, it will also be dissipating about 1W in heat! 1W out of a device that is absolutely tiny (3mm x 3mm) is an incredible amount of heat. The maximum absolute temperature of the device is 85C - at a temperature higher than this, permanent damage can occur. 

Ultimately the r9m's PCB thermal pad for the SKY65111 isn't optimised for the much higher duty cycle expressLRS pushes through it. Frsky's r9 firmware runs at a lower duty cycle, and Im sure crossfire runs lower too (not sure), but you could get away with no cooling and seemingly higher power if you keep things on for shorter periods if time! So then one wonders can the the duty cycle be dropped - but at what cost? Would you like to give up the performance enhancements that Express LRS offers? Didnt think so haha! No free lunches in life. 

So, there are a few design issues:
1. The thermal heat sink on the original r9m is just too small. 
2. the design choice to have the electronics on the top of the PCB means the thermal pad is nearly flush against the plastic backside... not only is the thermal pad small, it is next to plastic that would just trap the heat in! By the way the new happy model version doesn't fix this problem - its a huge heatsink but again next door to plastic! No wonder that thing needs cooling [Scroll to 3:37](https://www.happymodel.cn/index.php/2021/02/19/expresslrs-module-es915tx-long-range-915mhz-transmitter-and-es915rx-receiver/)

I believe a Fan-less version can be made, to do this

1. Select a higher efficiency power amplifier, or
2. Use a bigger thermal pad, or
3. move the RF amplifier to the underside of the board, and add a thermal heat sink that then fans out to outside the JR module to reject heat. 

This is a PCB thermal design problem that will continue to exist as long as designers continue to layout the SKY65111 the way they do with the thermal pad design that they do. ExpressLRS  users should be aware of this limitation and either limit power and try to remove heat with fans etc, or else risking packet losses, fail safes, or permanent damage to their transmitter modules. 

Attaching what I did to my r9m to manage this problem. Soldered a heat pipe directly to the SKY65111 thermal pad. The "heat pipe" was salvaged from an old computer that's why it looks banged up (Heat pipes are great for moving heat from one end into the other other). 

Will it go 1W without a heatsink? It will be an easy test. I think it will :D only one way to find out. 
[
[image]
[image]
](url)

Dropsleutel, 2021-10-02:
Nodoubt that it gets hot we know it.  But a hot amplifieron its own normally should not lead to a corrupt data link, but it lead to less power less range.

so the system gets hot >  then "some other component"  gets hot >  the the RF link gets corrupted.  (Possibly a X-tal VCOand link shifts frequency ? , or maybe it is the Frsky shitty invertor transistor getting out of wack, data gets not true when hot etc. Maybe the it is the semtech.    I can remeber that JamesK mentioned at the time that the over the air link still was ok, but the CRSFlink between Tx and R9M get corrupted.    Indeed cooling helps, but it is not the real root solution

brandonrc, 2021-12-09:
@token9 that is just insane what you did. 

@Dropsleutel what do we hope to get from this issue? I think we know that R9 hardware gets hot and can cause issues with link quality and even causing failsafe's. 

I am closing this issue because it is a hardware design issue and not a ELRS software issue.

If there is an issue with ELRS code causing this problem please feel free to open this back up. 

https://www.expresslrs.org/2.0/hardware/fan-mod/

I cannot speak for the rest of the dev's but using R9 above 250mw is dangerous and requires modifications and even then we cannot guarantee its use at those power levels. 

I recommend using dynamic power and using the lowest power required for the flight.

closed 2021-12-09 by brandonrc

Higirich, 2022-05-23:
Hi, 

I know it is a closed topic but I would like to ask your advice. 

What I noticed about my R9 system is the following:

If I fly 1.3 km I am starting to experience some data loss even when the antenna is in the best range. 

- (100 mw output power) EU version 868Mhz FW: 2.5 

Issue noticed: 

I did not used the module at 1W even with original firmware. 

I cheked the db numbers at home and I did not find any kind of issues. (Ref.: JB youtube chanel)

Required range: 

I would like to fly 2-3km of range. 

I have a feeling the module itself is under performing for some reason. (I printed a new antenna holder and did the best orientate the antenna)

Question: 

I would like to swith to a new module wich was designed for the ELRS but before doing that: 

Is it safe to run the R9 without mods at 250mw?