ExpressLRS/ExpressLRS #430: [Feature Request] Sharding channels to reduce the latency to 2.5ms or even lower

qdrk, 2021-04-02:
The idea is simple, since the ultimate latency limiting factor is the packet size. Let's cut the packet size by, say, 50% by using two pairs of SX1280s:

* 1 pair responsible for Throttle, Yaw, 2 aux
* 1 pair responsible for PITCH, ROLL, 2 aux

If the original latency for 500Hz is 5ms, this in theory could lower it to 2.5ms. 

This'll obviously need some changes on both ELRS and Betaflight, wonder how much change would it be for folks who are familiar with the codebase.

brandonrc, 2021-12-20:
@qdrk I know this is an old issue but looks like there was not much traction on the idea. I think it is interesting but something tells me this kind of this is unlikely to happen without someone doing most of the work. 

@CapnBry brought up an issue with not getting both packets.

brandonrc, 2021-12-20:
Another factor to consider, @wvarty brought up, ETX and BF do not support 1000hz so there is no real reason to use dual sx1280's right now.

qdrk, 2021-12-20:
Sounds good to close, higher frame rate would also come with more noise anyway, hard to tell if it's really going to make a difference in real flying.

closed 2021-12-20 by qdrk