ExpressLRS/ExpressLRS #446: Can't flash ES915RX via passthrough

sfjuocekr, 2021-04-08:
I'm getting the following error when I try to flash the receiver:

`Uploading .pio\build\HappyModel_RX_ES915RX_via_BetaflightPassthrough\firmware.bin
  ** Searching flight controllers **
      > FC found from 'COM3'

Detected the following serial ports on this system:
  COM3

=================== FIRMWARE UPLOAD ===================
  Bin file '.pio\build\HappyModel_RX_ES915RX_via_BetaflightPassthrough\firmware.bin'
  Port COM3 @ 420000
======== PASSTHROUGH INIT ========
  Trying to initialize COM3 @ 420000

Attempting to detect FC UART configuration...
    ** Serial RX config detected: 'serial 5 64 115200 57600 0 115200'
Enabling serial passthrough...
  CMD: 'serialpassthrough 5 420000'
======== PASSTHROUGH DONE ========

We were already in bootloader

uploading 44476 bytes...
ERROR:xmodem.XMODEM:send error: expected ACK; got b'C' for block 1`

The last line repeats a few more times before it bails.

Initially I tried to flash it on Linux, the python script wouldn't detect the serial config so I set it up manually and tried it just to get the same error. I went to Windows to double check my findings.

sfjuocekr, 2021-04-08:
I dusted off my STLink and flashed it that way, it works now :)

sfjuocekr, 2021-04-09:
I found the culprit, my TX on the FC isn't working as intended!

closed 2021-04-09 by sfjuocekr

rexfpv, 2022-01-26:
Hi there, I have the exact same issue but no acces to an st link. Can you elaborate on what exactly was the culprit, I’ve been at this for days and any help would be truly appreciated!

sfjuocekr, 2022-01-27:
The problem was something with the serial port on that FC, maybe if you connect it to another it might work?

rexfpv, 2022-01-27:
Ok thanks, I’ll try some other UARTS now and chime back in a bit