ExpressLRS/ExpressLRS #471: Feature: Add support for 6 position switch to Hybrid switch mode

brad112358, 2021-04-14:
I see there have also been a few requests on Discord for this...
I'd like to propose a change to GenerateChannelDataHybridSwitch8() and UnpackChannelDataHybridSwitch8() to add support for a 6 position switch (3 bits) on ch5 (switchIndex 1).  If I'm not mistaken, this can easily be done by using the unused switchIndex of 0 as the extra bit.

If I don't hear any objections, I'll go ahead and work up a pull request.  This may take a few days as I'm still new to ExpressLRS and I need to get my brand new hardware flashed and hook up a bench test setup so I won't feel bad if someone else implements this fairly obvious idea before I finish.  This should be very minimal impact and I'd love to see it make the next release.

JyeSmith, 2021-04-14:
You may have seen it, but if not check out the below discussion.

Currently we use 2b to indicate which switch option is being used.  So we can have up to 4 different options, but currently only have the 2.  I think a 3rd 8 X 8 way switches will give everyone the biggest bang for their buck and make the most use out of the available bits.

We would be very happy for you do contribute.  Make sure you're on discord if not already :)

https://github.com/ExpressLRS/ExpressLRS/issues/352

brad112358, 2021-04-14:
Thanks @JyeSmith,  I had searched for discussion both here and on discord, but somehow I missed #352.  I do think it would be interesting to steal some bits from yaw to create a new options with a few slower 8 or even 6 bit channels.

But that's not what this issue is about.  When reviewing the code in preparation for some more extensive work, I noticed that the current HYBRID_SWITCHES_8 code is effectively leaving one bit unused because switchIndex==0 is never generated or used.  This means that by assigning the same channel to both switchIndex ==0 and switchIndex==1, we can get an extra bit by keying off of which Index was used.  I'm proposing using this extra bit as the high order (or low order, I haven't decided yet) bit of ch5, giving us 8 possible values for this switch which meshes nicely with modern radios which have one 6 position (push buttons) switch.  The other 7 low-rate switches would be left at 2 bit each.  For FC firmware like Ardupilot which can have lots of different flight modes, it's confusing to have to use multiple switches for flight modes, but the 6 position (wish it was 8) switch works great.  Since we don't change any other behavior by converting one of the low rate 2 bit channels to a 3 bit channel, via this method, it seems like a win.  I hope this is more clear than I must have been in my initial comment.

brad112358, 2021-04-15:
@CapnBry, I can't keep up with the general channel on discord, but I just noticed you discussing something like this there.  I don't want to do further work on this if you are already about done.  Is your approach the same as I describe above?

JyeSmith, 2021-04-15:
Yep.  Keep an eye out for the pr soon for testing 👍

closed 2021-04-22 by AlessandroAU