ExpressLRS/ExpressLRS #499: Please give it a thought to use internal microcontroller STM8L15xG4 on 915MHz, 30dBm, SX1276 based LoRa module to be an ultimate low cost solution for Express LRS

VU2ATX, 2021-04-20:
While searching LoRa modules on the internet, I came accross a 915MHz 30dBm SX1276 based LoRa module (Part No: E32-915T30D) by a company called EByte.

This also has same LoRa chip as that of Express LRS and has onboard STM8L15xG4 host microcontroller and the board has SWIM programming port outside. Any cheap SWIM programmer like ST-Link V2 which costs less than 5$ can be used to program.

Rather than using additional ESP01, it should be thoughtfull to use this internal microcontroller.
Please do not mind if its a foolish idea.

VladimirP1, 2021-04-25:
This would require a rewrite of ELRS from scratch because there exists no C++ compiler for stm8 (except IAR I think which is not an option because of its cost). So the question is, is this worth the effort? Also, it seems this microcontroller is kinda slow, isn't it? Can it reliably parse CRSF at 400k baud?

VladimirP1, 2021-04-25:
Also, it seems that E32 is quite a crappy module actuallty. It does not have a high quality low-pass filter after the PA, and gives horrible interference to 5.8Ghz video link (A lot of QCZEK users are complaining about that and the E19 module has the same problem). So it will need an external low-pass filter.

VladimirP1, 2021-04-25:
There's one more problem with these E19 modules. They do not give you the ability to rapidly power the PA on and off (they burn out if you do). And if you do not power off the PA, the RX sensitivity drops dramatically, so you get almost not telemetry reception.

VU2ATX, 2021-05-02:
> Also, it seems that E32 is quite a crappy module actuallty. It does not have a high quality low-pass filter after the PA, and gives horrible interference to 5.8Ghz video link (A lot of QCZEK 868 users are complaining about that and the E19 module has the same problem). So it will need an external low-pass filter.

Yeah, that's really nice justification.

VU2ATX, 2021-05-02:
> This would require a rewrite of ELRS from scratch because there exists no C++ compiler for stm8 (except IAR I think which is not an option because of its cost). So the question is, is this worth the [...]

On searching on the internet, I found http://qczek.beyondrc.com/qczek-lrs-433mhz-1w-lora-rc-link/
This guy has done similar thing but has used internal microcontroller on the E32 module. I could find really several videos on YT using this for pretty long range. However this guy has kept the source closed atleast for now.

VladimirP1, 2021-05-02:
> On searching on the internet, I found http://qczek.beyondrc.com/qczek-lrs-433mhz-1w-lora-rc-link/
> This guy has done similar thing but has used internal microcontroller on the E32 module. I could [...]

This guy was writing his own firmware from scratch, he had the freedom to choose C or Assembly as his programming language. ExpressLRS has a lot of C++ code and I do not think it is feasible to translate in to plain C just for this STM8L.

The situation with toolchains for STM8 is very disappointing. There is Cosmic(free to use, very outdated) and IAR(costs a fortune), SDCC(free and open source, but the stm8 target has very basic functionality), STVD(assembly only, the ST's original dev environment).

So there's no Arduino on STM8 and there'll never be. Just because the SDCC compiler is too basic and the others'll have licensing problems.

And this means that to write on STM8 you have to do a lot of things in your own code, read really crappy datasheets (try understanding how to do an I2C transfer, hehe) and do a lot of debugging of basic things like I2C, timers, uarts, clock setup.

VladimirP1, 2021-05-02:
> > Also, it seems that E32 is quite a crappy module actuallty. It does not have a high quality low-pass filter after the PA, and gives horrible interference to 5.8Ghz video link (A lot of QCZEK 868 users are complaining about that and the E19 module has the same problem). So it will need an external low-pass filter.
> 
> Yeah, that's really nice justification though a Pi LFP costs around a $.

I know, I use lfcn-900 with my E19, this fixes video interference. But that's extra thing to buy, you have to get a filter like lfcn-900, a couple of dc-block caps, a couple of ufl connectors and a pcb to put all these things onto.

VU2ATX, 2021-05-02:
> This would require a rewrite of ELRS from scratch because there exists no C++ compiler for stm8 (except IAR I think which is not an option because of its cost). So the question is, is this worth the [...]

Yes, do agree that it need to re-write the code for STM8. I think it could handle CrossFire Protocol @400k baud.

closed 2021-12-20 by brandonrc