ExpressLRS/ExpressLRS #501: Feature: STM32WLE5JC support.

kifinnsson, 2021-04-21:
I am wondering if there is any interest in a port to the STM32WLE5JC? It may be overkill but if there is interest I'm willing to start a port to the LoRa-E5 mini board I also have the full LoRa-E5 Development Kit both from seeedstudio bazaar. 

This would be my first real dev work but if I can get ELRS running on it, curious if it's worth my time?

kifinnsson, 2021-04-21:
Links for reference:

LoRa-E5 Development Kit: https://www.seeedstudio.com/LoRa-E5-Dev-Kit-p-4868.html
LoRa-E5 mini: https://www.seeedstudio.com/LoRa-E5-mini-STM32WLE5JC-p-4869.html

JyeSmith, 2021-04-21:
crteensy on discord is also looking at this MCU.  Might be worth while jumping on the development chat and collaborating.

JBKingdon, 2021-04-21:
100% worth a try. The 1262 isn't supported yet but looks to have great potential as a better than current 915 link. I want to get the e22 boards (also 1262 based) supported for the same reason. The stm on the wle5 looks weak so may be challenging, but I think @cruwaller managed to get the rx running on a similar spec chip. Perhaps worthing ditching ardiuno for better efficiency?

cruwaller, 2021-04-22:
yeah, it can be supported but requires optimizations. Not a trivial task but can be done.

closed 2022-02-04 by github-actions[bot]