ExpressLRS/ExpressLRS #506: Feature Request - Test/Debug mode,

ballistRC, 2021-04-22:
It would be very useful on an RX if you could turn on the TX, change frequency and I believe tune the XO.  By being able to observe the TX output, it would prove major portions of the design working without needing end2end comms.  RxDumping of payload would also be useful.  I suppose this could also be added to the TX module firmware.  I would also be useful if say the XO tuning info was saved to config eeprom.
There are a number of other useful functions available in the SX12xx and exposing them would assist future R&D and fault finding.

AlessandroAU, 2021-04-22:
I agree with having a debug mode, by XO offset for both the sx1276 and the stm3/esp8285 is already compensated for

ballistRC, 2021-05-01:
I will make a feature list of what I think needs to be exercised.  It should talk xmodem to a terminal and have a simple command interpreter.  I could write the command interpreter and wrap it in a function but I am really rusty with C and have no idea how to communicate with xmodem or the application structure but I am willing to give it a go.  I'm a hw guy who spends all day writing system verilog and such.

brandonrc, 2021-12-12:
@AlessandroAU should we close this issue? Seems like it has been open with no development on  it.

closed 2022-01-21 by maybenikhil